mne-tools/mne-python #3740: AssertionError brainvision.pyc file

clementdonde, 2016-11-05:
Hello,

I can't open my .vhdr files for EEG analysis, there's got to be a pb with my brain vision.pyc file. Can someone guides me to fix this ?
Thxs in advance !
[image]

agramfort, 2016-11-06:
please share a problematic file.

how was this file created?

clementdonde, 2016-11-06:
Thxs for your answer.
The file was downloaded with the brain vision package.

[brainvision.pyc.zip](https://github.com/mne-tools/mne-python/files/573966/brainvision.pyc.zip)

agramfort, 2016-11-06:
I meant share the .vhdr file so we can replicate the error and ideally fix
it.

thanks

clementdonde, 2016-11-06:
ok, sure

clementdonde, 2016-11-06:
ok, sure
[exemple.vhdr.zip](https://github.com/mne-tools/mne-python/files/574093/exemple.vhdr.zip)

agramfort, 2016-11-07:
ok so you have data in ASCII and our reader only supports data in binary
format
(binary by the way more compact on disk and avoids loss in numerical
precision)

can you save them to binary so it fixes your problem?

See https://github.com/mne-tools/mne-python/pull/3748

clementdonde, 2016-11-08:
thxs for this quick solution.
do you have a method to access my .vhdr files content and translate it to binary ?

agramfort, 2016-11-08:
no.

but if you share a full dataset we can look into it.

you can't save again to binary with your brainvision software?

Eric89GXL, 2016-11-08:
Would it be possible for you to get a short segment (say, 1 sec?) of the ASCII data and upload it? It might also be possible to add support for it at some point.

clementdonde, 2016-11-08:
do you know which program other than my terminal can open vhdr file ? I can't open it with anything so can't get segments out of it. 
[Archive.zip](https://github.com/mne-tools/mne-python/files/578683/Archive.zip)
please find attached the vhdr + dat and vmrk corresponding files, if it can help.

thxs in advance

agramfort, 2016-11-08:
more news happening at https://github.com/mne-tools/mne-python/pull/3752 ...

clementdonde, 2016-11-08:
ok, nice job! 
I try to use this conversion tool but I'm a very beginner at MNE, can't find solution to solve this error, can you advise me ?
thx in advance

[image]

jaeilepp, 2016-11-09:
See instructions at http://martinos.org/mne/stable/contributing.html#some-other-things-you-might-want-to-do on how to fetch a single pull request. Once https://github.com/mne-tools/mne-python/pull/3752 gets fixed you should be able to read the ascii files with ``mne.io.read_raw_brainvision(fname)``.

Hope this helps.

clementdonde, 2016-11-27:
my terminal can't open the files even after conversion of the .vhdr file from ASCII to binary.
please find attached the error code
[error.pdf](https://github.com/mne-tools/mne-python/files/614646/error.pdf)

Eric89GXL, 2016-11-27:
How did you convert it? I don't think the header file was what needed to be
converted, but rather the data file.

Did you try my PR with your original ASCII data?

Eric89GXL, 2016-11-27:
(... And the conversion would need to be done with a brainvision tool,
which I suspect is not the case here)

On Nov 27, 2016 10:48, "Eric Larson" <larson.eric.d@gmail.com> wrote:

> How did you convert it? I don't think the header file was what needed to
> be converted, but rather the data file.
>
> Did you try my PR with your original ASCII data?
>

agramfort, 2016-11-27:
Yes I tried this PR with the original files cf my private email

Eric89GXL, 2016-11-27:
What error did you get on my PR with the original files? I thought I had
the original file working. Or is there a new / different failing file now?

Eric89GXL, 2016-11-27:
I had a little bug, should hopefully work now

PR #3752 (2016-11-28): MRG: Allow ASCII BV files
Closes #3740.

Still needs tests. Seems to work with data @clementdonde provided.
closed 2016-11-28 by agramfort