mne-tools/mne-python #3746: BUG: Over decimation because of units issue

kdoelling1919, 2016-11-07:
Hi all,
I'm loading in raw files from KIT system at NYU using `mne.io.read_raw_kit()` and found that the digitized head shape in `raw.info['dig']` was only producing 9 points (the 3 anatomical markers, five fiducials and 1 lone head-shape pt). This didn't use to be a problem so I tried loading old data which had worked fine before and now they have the same problem. The issue seems to be that units for the KIT head shape were changed from mm to m in `_read_dig_points()` in the meas_info.py file. This is fine unless you're like me and always have way too many points forcing automatic decimation in `_set_dig_kit()` in kit.py. That call looks like this:

```python
if n_pts > KIT.DIG_POINTS:
        hsp = _decimate_points(hsp, res=5)
```

I think the fix is that the 5 should be 0.005 now that we are in meters. Otherwise you have voxels with side-length of 5 meters which may be a bit of an over decimation. This fixes my issue at least. I am happy to make the PR but wanted to make sure that's not going to effect anything else that I'm not foreseeing. 
What do you guys think?
@teonbrooks @christianbrodbeck

Eric89GXL, 2016-11-07:
Sounds reasonable to me

christianbrodbeck, 2016-11-07:
Yes that sounds right.

Apart from implementing the fix, you could consider downsampling with FastScan, it has a better algorithm

PR #3747 (2016-11-22): FIX: Updating digitizer point decimation to units of meters
Updating automatic decimation of digitizer headshape to units of m rather than mm in keeping with recent updates. Fixes #3746 

@teonbrooks @christianbrodbeck @Eric89GXL
closed 2016-11-22 by Eric89GXL