mne-tools/mne-python #3764: Official docs still 0.12

christianbrodbeck, 2016-11-12:
I just noticed that http://martinos.org/mne/stable/ still points to the docs for 0.12

agramfort, 2016-11-12:
can you be more explicit?

christianbrodbeck, 2016-11-12:
The window title says mne 0.12 and I tried to look up a method signature
that changed from 0.12 to 0.13 (SourceEstimate.plot()) and it showed the
0.12 version (
http://martinos.org/mne/stable/generated/mne.SourceEstimate.html#mne.SourceEstimate.plot
ought to have )

christianbrodbeck, 2016-11-13:
[image]

agramfort, 2016-11-13:
arfff maybe it's because I pushed a fix to maint/0.12 branch and it updated
the git repo containing the web site :(

let's see if this fixes the pb

https://github.com/mne-tools/mne-python/commit/bdbbb84a46a20e7cdcc4ca08621a5c939791117c

Eric89GXL, 2016-11-14:
Should we start keeping old documentation versions like sklearn, scipy,
etc.?

Eric89GXL, 2016-11-14:
Now GitHub is correct, martinos is not

mmagnuski, 2016-11-14:
Would be really nice if there was some drop-down menu to switch between versions (for example). Similarily to how it works on readthedocs (Julia docs for example).

agramfort, 2016-11-14:
fixed

closed 2016-11-14 by agramfort

Eric89GXL, 2016-11-14:
Do we want a drop down with different doc versions? It would also fix this
overwriting problem.

agramfort, 2016-11-14:
why not but it's not a priority for me