mne-tools/mne-python #3769: [Question] CSP as the solution for one category and the sum instead of two categories?

noreun, 2016-11-15:
Hi MNErs,

First I apologize if this is a trivial question, but I'm struggling to find more information about that.

I've been performing some classification in my data using the CSP class, and noticed that instead of solving the generalized eigenvalue problem for the covariance of category A and covariance of category B (covs[0] and covs[1]), as implemented in all three references:

Zoltan et al., 1990; Blankertz et al., 2008.; Grosse-Wentrup et al, 2008.;

the method CSP.fit is solving it for covs[0] and covs.sum(0)
 
            eigen_values, eigen_vectors = linalg.eigh(covs[0], covs.sum(0))

I also noticed until this commit about one year ago:

"Simplified CSP algorithm"
https://github.com/mne-tools/mne-python/commit/658fb33cea6ed3a45927c02927303712b1a917b4

the algorithm was indeed solving the problem for covs[0] and covs[1] (or cov_a and cov_b)

**So my question is: are there some references showing the differences between the two approaches, and/or when they are equivalent?** 

I compared the results and they are not the same, so I assume there must be some extra step normalizing/transforming the data if they are to become the same.
And if indeed they are not, what do we lose, when / why perform this simplified version?

Intuitively I would think that by solving the problem for A and the sum of A and B, we are introducing **some** variance of A into the second matrix, and therefore decreasing the original difference between the two variances. This would make it harder to find a solution for the generalized eigenvalue problem when compared to A vs B. Does this makes sense?

Is there some simulations / comparisons of the two methods?

Best Regards,
Leonardo

jasmainak, 2016-11-15:
Please take a look at this PR: https://github.com/mne-tools/mne-python/pull/2630 and the related issue: https://github.com/mne-tools/mne-python/issues/2437. Report back if you have any questions after that.

noreun, 2016-11-15:
Thank you @jasmainak , I assumed there was no issue related to the commit since normally there is a mention to the issue number in the message. Sorry about that!

All my questions where indeed answered in #2630, I'm closing this issue.

One last question, is someone working in the variants of CSP that uses spectral information as mentioned in in #2437 (specially CSSP)? This is the only post I could find in github about that, and I'll probably need to implement something like that soon.

agramfort, 2016-11-16:
if someone has the time to implement these CSP variants with test and an
example

I am all for it !

larsoner, 2023-10-01:
Nowadays let's go to discourse for discussions/questions, see e.g. https://mne.discourse.group/t/cssp-and-other-advanced-algorithms-for-bci-decoding/3359 . And if someone is motivated to add a specific version to MNE let's open a new issue about it!

closed 2023-10-01 by larsoner