mne-tools/mne-python #3781: more consistent class instances/types

drammock, 2016-11-21:
It is unintuitive and surprising (at least to me and @jasmainak) that:

```python
my_epochs = mne.read_epochs('my_data-epo.fif.gz')
isinstance(my_epochs, mne.Epochs)  # returns False ?!!
```

It would be nice if `mne.epochs.EpochsFIF` were a subclass of `mne.Epochs` so that type checking were more straightforward.  Similar argument applies to the various different types of `Raw` objects.  @Eric89GXL has suggested (on Gitter) making `_baseEpochs` public as one solution, with a caveat in the docstring that it shouldn't be instantiated directly in user code.

jaeilepp, 2016-11-22:
I'm wondering how much damage it would cause to rename current ``_baseEpochs`` to ``Epochs`` (and make it public) and the current ``Epochs`` to something else like ``EpochsRaw``. I feel this is how it actually should be.

jaeilepp, 2016-11-22:
Of course, now I realize that the way we construct the epochs is through ``mne.Epochs``, so it seems painful.

jaeilepp, 2016-11-22:
How do you folks feel about having the construction of epochs being done through a method of raw instance? It would be kind of in line with how we create evoked (``epochs.average``). So the API would be something like ``raw.make_epochs(events, event_id, ...)``. Same for goes for ``raw.find_events(...)``.

This way we would work around this issue and be able to eventually rename the ``_baseEpochs`` to ``Epochs``.

jona-sassenhagen, 2016-11-22:
We discussed this before (a `raw.to_epochs` method). A few guys (I think @jasmainak , maybe @teonbrooks ..? and I) were very much in favour.

jasmainak, 2016-11-22:
@jaeilepp you do not want to [reopen old issues](https://github.com/mne-tools/mne-python/issues/2872) :)

what do you think of simply adding an `epochs_types = _BaseEpochs`? That way it is clear that it is meant to be used only with `isinstance` and nothing else. Maybe I'm being naive about something, as API discussion is never simple ;)

jaeilepp, 2016-11-22:
That's one option. My suggestion was more of a long term vision of eventually moving towards that direction. There is a logical disconnection of having ``EpochsFIF`` that does not inherit ``Epochs``. Also, we don't really have any other classes that are constructed directly by calling the constructor (apart from the array constructors).

drammock, 2016-11-22:
> There is a logical disconnection of having `EpochsFIF` that does not inherit `Epochs`.

This is exactly my view as well. Would it work to make the array constructor serve as the base `Epochs` class,  and have any other epoch classes inherit from it? To me that's the most logical arrangement... The default case is the case where your data are already in numpy format, and the special cases are direct-to-epochs from some other format, saving you the trouble of manually converting to numpy first.

Eric89GXL, 2016-11-22:
> Would it work to make the array constructor serve as the base Epochs class, and have any other epoch classes inherit from it? To me that's the most logical arrangement... 

In general we don't want to force people to have things in memory. `EpochsFIF`, `Epochs`, and (maybe?) `EpochsKIT` all allow on-demand reading, as do all of (or almost all of) our `Raw*` classes. So in this sense, thinking about objects as thin layers over `numpy` arrays / data in memory isn't great.

There is also a historical issue in that `Epochs(raw, ...)` has been *the* way to construct epochs for a long time. Probably the two most used / most important / core classes in `mne` are `Raw` and `Epochs`. Lots of people's code thus uses this pattern, and asking them to change does not seem worth it to me for such a fundamental object. Sure it's a bit of a misnomer, but so far this seems to cause fairly minor inconveniences.

Can people live with `epoch_types`? I'd much rather deal with this through documentation and minor changes than an overhaul of our fundamental types.

agramfort, 2016-11-22:
+1000 !

or just do

isinstance(epochs, _BaseEpochs)

as we have a base class for Epochs

Eric89GXL, 2016-11-22:
In that case we can rename `_BaseEpochs` -> `BaseEpochs` to avoid people needing to use private vars. That should work okay, especially if coupled with some documentation in `Epochs` and `read_epochs` about what classes are produced. PR welcome if someone has time.

drammock, 2016-11-22:
Honestly `epochs_types` strikes me as a bit of a band-aid solution... we're not actually dealing with a transition in a dependency's API here (as in the py 2-to-3 transition), we're talking about consistency *within* mne. That said, I don't have a ready-to-hand idea about how to overhaul the class system without it being a ton of work. But I do believe that it should be done. Maybe next GSoC?

Eric89GXL, 2016-11-22:
I am -1 on restructuring it. If I were creating MNE from scratch, I would structure it differently. I agree it's not obvious or consistent. But the downsides of restructuring fundamental classes is not worth it. We have too many users and too much existing code to make the switch.

drammock, 2016-11-22:
> Lots of people's code thus uses this pattern, and asking them to change does not seem worth it to me for such a fundamental object.

sklearn did something similar a while back, and one of the goals of that change was consistency. In my view these kinds of overhauls are part of the normal pre-version-1.0 throes of software development.

agramfort, 2016-11-22:
let's say you have a make_epochs function that takes a raw instance like
Epochs constructor now.

how will it help you to do type checking?

drammock, 2016-11-22:
@agramfort I agree that a function `mne.make_epochs(raw, ...)` does not address the type checking issue.  Neither does a `raw.to_epochs` method that @jaeilepp suggested.  But I definitely don't agree with your suggestion to use the private class and 

> just do `isinstance(epochs, _BaseEpochs)`

@Eric89GXL's suggestion to make `epochs_types` would solve the problem, but has drawbacks that I've mentioned above.  The *simplest* solution to the type-checking problem is making `_BaseEpochs` public (and adding appropriate warnings if people try to use it directly as a constructor).  But the *best* solution, in my opinion, is to have a base class called `Epochs` that is used as a constructor, and triages to create an object of one of the subclasses (`EpochsArray`, `EpochsRaw`, `EpochsFIF`, etc) depending on the type of its first argument.  All the subclasses inherit from the base class, and the base class is public (for easy type checking) and (as a bonus) the base class gets the most generic name, "Epochs".

jasmainak, 2016-11-22:
@drammock let's not try to overfit and be perfect. Perfect is the enemy of good (and also done) :) These things are not obvious when you start a library and it is only apparent in retrospect. We should be a bit more patient before suggesting an overhaul. There are many more urgent things to be done than a class overhaul for a GSoC. I'm sure we can live with minor inconveniences.

Among the suggestions so far, I like my own `epochs_types` of course -- because it has a ring of familiarity and also because it offers a public API without exposing the entire class. But I can also live with making `_BaseEpochs` public if this is what everyone prefers. Please make a PR if you are up for it!

jasmainak, 2016-11-22:
@drammock what is also perhaps escaping your attention is the `RtEpochs` class. I would encourage you to play around with it and then report back ;)

mmagnuski, 2016-11-22:
I like `epoch_types` but maybe extend it a little to `mne_types['epochs']` (so that one can also test `mne_types['raw']`) - multiple time when I had to perform type checks on mne objects I had to check where to import the types from or ensure myself if I remember the base type name correctly etc. `mne_types` would be a convenient solution.

drammock, 2016-11-23:
@jasmainak I don't see how this request constitutes "overfitting".  I've offered what is clearly labeled as an opinion about what I think is unintuitive about the current design and how I think it could be improved, and I've clearly acknowledged that it's hard and need not be done right away.  Saying you prefer the `epoch_types` solution is fine, but saying "perfect is the enemy of the good" implies that you don't want it done right at all, and that the band-aid solutions suggested here are in fact preferable to what I've suggested (i.e., the status quo bias mentioned on #2872).

jasmainak, 2016-11-23:
> @jasmainak I don't see how this request constitutes "overfitting"

because you want to change the class structure for one use case.

anyway, I'm not the one you have to convince. It's @agramfort and @Eric89GXL :) I recognize that it's a valid use case, but I think we need to be more patient before suggesting more radical solutions. I'm probably wiser now ;)

I'm sure a PR for the "band-aid" solution or making `_BaseEpochs` public is still welcome!

jaeilepp, 2016-11-23:
>let's say you have a make_epochs function that takes a raw instance like Epochs constructor now. how will it help you to do type checking?

I would imagine if we had a convenience function of ``raw.to_epochs``, which would save you quite a bit of writing (``epochs = raw.to_epochs(event_id=1)`` instead of ``events = mne.find_events(raw), epochs = mne.Epochs(raw, events=events, event_id=1)``). Eventually, people would convert to using the method instead of using the constructor. This means that changing the naming through couple of deprecation cycles would become less painful. Looking at the code now, I would even prefer having ``raw.to_evoked`` convenience function.

jona-sassenhagen, 2016-11-23:
ping @dengemann , who was really opposed to to_epochs last time :)

agramfort, 2016-11-23:
the more compact you make the notations the more you make it magic.

by doing

events = mne.find_events(raw)
epochs = mne.Epochs(raw, events=events, event_id=1))

you can show to users what events are. You can edit / correct them which
is often needed. I am afraid of solutions that work out of box for 50% of the
data and is nightmare for the other half. I would prefer a solution that is 
more verbose but it easier to have it working for 90% of the data.

my 2c

dengemann, 2016-11-23:
I keep being -1000 against `.to_epochs method`. The way of constructing
Epochs is not broken and does not need a fix and is a robust and well
working routine.
As @agramfort says we don't need more magic here and it is good force users
to know explicitly how they want to construct epochs.

jona-sassenhagen, 2016-11-23:
> On 23 Nov 2016, at 16:58, Denis A. Engemann <notifications@github.com> wrote:
> 
> I keep being -1000 against `.to_epochs method`. The way of constructing
> Epochs is not broken and does not need a fix and is a robust and well
> working routine.
> As @agramfort says we don't need more magic here and it is good force users
> to know explicitly how they want to construct epochs.
> 

To be fair, the issue of "magic" is orthogonal to the issue of adding a method; the method could in principle be set to require the same input as the current constructor (minus the Raw instance), and we have recently simplified the current constructor to perform some magic.

dengemann, 2016-11-23:
Guys I don't understand why we need to discuss proposals again and again
which do not improve functionality but rather mere aesthetic needs or
wishes for symmetry.
To me, we don't want to convert Raw to Epochs. It is a welcomed disruption
that you wait a second before you construct epochs.
Why the heck do we need to change the API of such fundamentally tested
things again and again.
These are some parts of MNE that I would like to be able to teach in 2020
as I taught them in 2013.
Wer are writing scientific academic software for which stability and
reliability matters, this is not an App Store.

drammock, 2016-11-23:
@jasmainak you say this is "overfitting" 
> because you want to change the class structure for one use case.

This is not about my use case. `isinstance(my_obj, SomeClass)` is *the* standard pythonic way of type checking, and you (and others) are suggesting to depart from that.  More fundamentally, if you're going to write software that makes heavy use of classes and subclasses, it is my opinion that, at the very least, the base class should be public to allow type checking against it (so that users don't have to write `isinstance(some_epochs_obj, [mne.Epochs, mne.epochs.EpochsFIF, mne.epochs.EpochsKIT, ...])`).  So far I have not heard a single refutation of this idea.

Oh, and by the way @jasmainak if you're not the one that needs to be convinced, please stop encouraging people to "just go make a PR for it".  The purpose of issue discussions is to hammer out these decisions BEFORE anyone wastes time implementing a solution that isn't going to get merged by the gatekeepers anyway.

@dengemann, @agramfort it is clear that there is still some baggage left over from the discussion of #2872.  Can we please agree that this issue is about type checking and class structure and NOT about the `raw.to_epochs` method?  @jaeilepp suggested it as one possible way of easing the transition for users during a major class overhaul, but I think it's painfully clear that there is absolutely no support for a class overhaul anyway, and there are two other candidates on the table (make `_BaseEpochs` public and make an `epochs_types` variable to check against).

agramfort, 2016-11-24:
I would make _BaseEpochs public so you can do

isinstance(epochs, mne.epochs.BaseEpochs)

works for you?

drammock, 2016-11-25:
@agramfort yes, that works.  I'll try to have a PR opened by monday.

dengemann, 2016-11-25:
Good idea. My support!!

PR #3794 (2016-11-27): MRG: Public base classes
closes #3781 

`make test-fast` passes locally; due to the nature of the changes (almost entirely find-replace) I didn't bother with `make test-full`.
closed 2016-11-27 by agramfort