mne-tools/mne-python #3784: Error using make_bem_model: inner and outer skull surface cross

romquentin, 2016-11-22:
Hi, 

I got this error on almost half of my participants after running `make_bem_model`:
`Surface inner skull is not completely inside surface outer skull`

I checked on freeview and it is true :) (I checked also pial and white surface which seem good).

Do you know how to correct that?

Thank you

kingjr, 2016-11-22:
It's generally because of a signal dropout in the bottom slices of the T1 MRI. You should be able to fill it with 1.

romquentin, 2016-11-22:
with 1?

kingjr, 2016-11-22:
any value that is different from 0.

romquentin, 2016-11-22:
you mean directly on the inner and outer skull surface or on a previous image and I need to re-run the make_scalp_surface? Not sure how to fill with 1...

kingjr, 2016-11-22:
I think you have to modify the actual T1 and relaunch free surfer. @Eric89GXL do you know if the dropout is fixable at the BEM level?

Eric89GXL, 2016-11-23:
I have done the fill-with-value fix to make it so that the head surface can be generated, but never so that BEM surfaces don't intersect. I'm not sure if it would help or not. 

But let's pick up the discussion over at https://github.com/mne-tools/mne-python/issues/3291 since I think it's the same issue

closed 2016-11-23 by Eric89GXL