mne-tools/mne-python #3786: `len(epochs)` on preloaded data

mmagnuski, 2016-11-23:
I have preloaded epochs that have been read from a preprocessed eeglab file - so they're already clean (I don't need to drop anything).
However, `len(epochs)` tells me:
> Since bad epochs have not been dropped, the length of the Epochs is not known. Load the Epochs with preload=True, or call Epochs.drop_bad(). To find the number of events in the Epochs, use len(Epochs.events).

Is it necessary for preloaded epochs? The check [here](https://github.com/mne-tools/mne-python/blob/master/mne/epochs.py#L1134) does not tests for `epochs.preload` while the error message suggests that it should.
Who is wrong in this case - me or mne? :)

(the `len(epochs)` is actually performed in `TimeDecoding` so I have to conform to what `len(epochs)` wants)

jaeilepp, 2016-11-23:
I believe ``preload_at_end=True`` should be passed to ``_BaseEpochs`` since the non-preloading is not supported for ``EpochsEEGLAB``, right? I've never used EEGLAB files, so I'll let you guys sort this out.

mmagnuski, 2016-11-23:
@jaeilepp - yes, IIRC `EpochsEEGLAB` should be always preloaded.

jasmainak, 2016-11-23:
oh no, you can have `preload=False` for EEGLAB files. Sometimes you have an `.fdt` file, so you can read it in chunks ...

jaeilepp, 2016-11-23:
@jasmainak so how do you read those with ``preload=False``? I don't see a ``preload`` argument for ``EpochsEEGLAB``.

jasmainak, 2016-11-23:
oh, we're talking about epochs. I'm still in my French class :) you're right.

So, `preload_at_end` does what? Perhaps I'm missing why `preload` should matter when we do `len(epochs)` ...

jaeilepp, 2016-11-23:
> So, preload_at_end does what?

Loads the data. In this case the problem is that preload flag is not set when it should be.

mmagnuski, 2016-11-23:
From my point of view the issue is rather that `__len__` does not check `epochs.preload` (which returns True in my case) but kind of assumes it is not if epochs were not rejected.
The notes for `__len__` mention that *This function only works if bad epochs have been dropped.*, but I don't get why it shouldn't work if data are preloaded but bad not dropped (especially if there are no bads)...

jaeilepp, 2016-11-23:
Well you can always do ``len(Epochs.events)`` which shows what you want (as the error message suggests). I think it's quite nice to have it only return the good epochs. Otherwise it can be a bit misleading.

jaeilepp, 2016-11-23:
Okay, it seems that it's the ``_bad_dropped`` flag that gets not being set. I suppose you can just set it to True when reading the file, since the constructor doesn't have rejection parameters available. Normally it gets set when you read in the data.

mmagnuski, 2016-11-23:
The issue is not strictly that I cannot do otherwise, but that TimeDecoding checks `len(epochs)`.
I can do `epochs.drop_bad()`, I just didn't think it was necessary if all my epochs were good (preprocessed in eeglab).

mmagnuski, 2016-11-23:
@jaeilepp: Ok, I get where it comes from now. I was setting `epochs._bad_dropped` to True myself but though it's not the way the world should work ;)

jaeilepp, 2016-11-23:
Yes. PR welcome.

PR #3793 (2016-11-25): FIX: set _bad_dropped to True after constructing epochs data from eeglab file
fixes #3786
currently I'm setting this manually - this may be not be the best way to do this, but passing `preload_at_end` to _BaseEpochs constructor does not work in this case - the data are already preloaded and passed to the constructor while `preload_at_end` asserts that `_data` is None.
Let me know if there is better way to do this - but it seems to work well currently.
closed 2016-11-25 by agramfort