mne-tools/mne-python #3787: FIX & ENH: linear_regression_raw

kingjr, 2016-11-23:
The solver parameter in `mne.stats.linear_regression_raw` can now be set with a function. However, the `y` is transpose as compared to the expected equation `w = X^T Y`

Here is  the copy from the default solver used in `linear_regression_raw`:
```python
linalg.solve(a, X.T * y.T, sym_pos=True, overwrite_a=True, overwrite_b=True).T
```

For example if one use a sklearn's ridge, we would have to transpose y before passing the solver:

```python
from sklearn.linear_model.ridge import ridge_regression
solver = lambda X, Y: ridge_regression(X, Y.T, alpha=1.)
event_id = dict((c, ii) for ii, c in enumerate(categories))
evokeds = linear_regression_raw(raw, events, event_id, solver=solver)
```

I suggest a couple of things:
* transpose `Y` back to its original shape
* detect whether the solver has a 'fit' attribute, in which case it'd do `solver.fit(X, Y).coef_`
* add a dictionary to call the commonly used regressions (e.g. `solver='ridge'`)

jona-sassenhagen, 2016-11-23:
It seems everyone who uses this uses sklearn's Ridge anyways, so it would be good to be more sklearn compliant and have a dict path to Ridge here.

I will do this.

kingjr, 2016-11-23:
Can you also defaults  `event_id` to `dict((k, v) for ii, v in enumerate(np.unique(events[:, 2])))` in that case ?

jona-sassenhagen, 2016-11-25:
Hm. Seems fine to me I guess.

@Eric89GXL would we need a deprecation cycle here? Currently, the solver is private, and the default use case does not change, but user-defined solvers will need an additional `.T`.

(I assume the number of affected users is super low though, and I probably know 90%+ of them personally ...)

jona-sassenhagen, 2016-11-25:
I'd also like to make event_id optional, I barely even use it anymore.

Eric89GXL, 2016-11-28:
> would we need a deprecation cycle here?

If it's a bugfix then no. If it's just for API reasons then yes.

> linalg.solve(a, X.T * y.T, sym_pos=True, overwrite_a=True, overwrite_b=True).T

Just saw this. Keep in mind that `ndarray.T` is a view, and thus allowing overwriting the contents could mean that they get destroyed, so make sure that users know that the `X` or `Y` they pass could be modified inplace.

Eric89GXL, 2016-11-28:
...although I guess you could check sizes and tell by the dimensionality, and warn about the new style? The only time you'd be in trouble is when Y is square, which might be rare enough to live with

jona-sassenhagen, 2016-11-28:
>The only time you'd be in trouble is when Y is square, which might be rare enough to live with

Not only rare, but also basically pointless - that'd mean more channels than time points, in which case you really shouldn't be doing rERP :)

>it's just for API reasons then yes.

It's just for API reasons. When I wrote the function, I wasn't thinking about sklearn.

But I guess if we transpose y based on its dimensionality, it'd be sufficient to just warn for 2 releases or so.

kingjr, 2016-11-28:
+1 for dim check + deprecation

jona-sassenhagen, 2017-03-28:
Q: maybe we should *error* when the array has the wrong shape. Rationale: factorizing/inverting/... a matrix with the wrong dimensions can lead to massive arrays that overload RAM and might freeze your Python session/computer. Thoughts? @Eric89GXL @kingjr

Eric89GXL, 2017-03-28:
I think it's better to be stringent / make sure the user passes the right thing, so +1 for error

jona-sassenhagen, 2017-04-12:
Close for #3857

closed 2017-04-12 by jona-sassenhagen