mne-tools/mne-python #3788: add events parameters to epochs.plot method

agramfort, 2016-11-24:
sometimes you want to plot events to top of epochs in epochs.plot.

an easy way is to add an events param to epochs.plot

jaeilepp, 2016-11-24:
It might not be that easy, since the time information is lost when the data is epoched, right? So passing the same events that created the epochs would not make sense unless you compute the offset between the events. Perhaps it would be easier/cleaner to include the stim channel to the epochs and then get the data from there.

The param for ``epochs.plot`` could then be either bool, int or list. So that True plots all the events, int plots only the event with that id and the list could have multiple ids.

hoechenberger, 2016-11-24:
This would indeed be an extremely useful addition. Once in a while events get "lost" during recording for whatever reason; the ability to plot all event markers on epoched data would help with visual inspection there.

bburle, 2016-11-24:
> It might not be that easy, since the time information is lost when the data is epoched, right?

Indeed, and this is a strong limitation in my view. Just take a simple RT experiment: once epoched on the stimulus, you loose the response (one trick is to save the RT in the second column of the "events" structure, but this is a hack, and not generalizable to more events). It would be useful to keep the events that do belong to the epoch.

> So passing the same events that created the epochs would not make sense unless you compute the offset between the events. 

Not sure! I believe that the onset of the events in the epoch object actually keep the absolute position in the raw, so the other event could be kept in real position.

Eric89GXL, 2016-11-24:
> It might not be that easy, since the time information is lost when the
data is epoched, right?

It is in epochs.events

> So passing the same events that created the epochs would not make sense
unless you compute the offset between the events.

I think you could compare epochs.events and the events array passed into
the plotting function to determine where the lines go, yeah. I don't see a
problem with this?

mmagnuski, 2016-11-24:
I agree with @bburle and @Eric89GXL - one can use `epoch.events` to get information about epoch location with respect to raw, and select relevant rows from raw `events` array and change their original latency to 'epoch-related' and plot. It's kind of backward engineering, but it is possible. :)

When it comes to things like loosing RTs etc. - I often use pandas for more complex experiments - so that I have for example one row per epoch with whatever variables I want. And that is something that would have to remain on the user side with how Epochs work in mne.

closed 2016-12-07 by agramfort