mne-tools/mne-python #3790: FIX/ENH: more generous EDF column name handling instead of hex-edits

dengemann, 2016-11-24:
One of the best moments today during the MNE workshop in Ghent: some participant is fixing his EDF file by hex-editing non-unique channel annotations . . . 
It made me think we should have some more generous mechanism in the EDF readers that raise a warning and disambiguate double column names by appending a running number to their name and giving the same message as now as a warning instead of throwing an error.
We speculated that unused columns may have identical default names.

Any thoughts @Eric89GXL @teonbrooks @jaeilepp @jona-sassenhagen ??

agramfort, 2016-11-25:
can you get your hand on one of these files?

ddumalin, 2016-11-29:
I'm the participant in Ghent that hex-edited his EDF files.
Please, find a zip file with 2 files: (1) sample_original.edf and sample_hexedit.edf.
I changed the 4 duplicate column names "EDF Annotations" to "EDF Annotation[1:4]"

[Sample_EDF.zip](https://github.com/mne-tools/mne-python/files/619166/Sample_EDF.zip)

agramfort, 2016-12-01:
thx @ddumalin for sharing.

@jaeilepp please have a look when you get the time. thx

closed 2016-12-14 by agramfort