mne-tools/mne-python #3801: Viz improvement: AverageTFR.plot_joint

jona-sassenhagen, 2016-11-29:
[image]

This is something EEGLAB has. I dislike normal TF plots ... these are much nicer IMO. Thoughts?

It's not *that* much code, and the most complicated part is already in `evoked.plot_joint`.

@jaeilepp @kingjr

jona-sassenhagen, 2016-11-29:
@jaeilepp don't make a PR already, let me try first :D

jaeilepp, 2016-11-29:
Sure, go ahead.

mmagnuski, 2016-11-29:
It may be nice if the channels (set of channels) used for the TFR is highlighted in the topomaps (although the user will always know which channels the TRF is obtained from...).
Would be also cool if topomapped parts of the TFR could be set in a rectangle way (for example in time, Hz order: `[(-0.2, 0), (8., 10.)]`) - but that could be something for later. :)

jaeilepp, 2016-11-29:
> Would be also cool if topomapped parts of the TFR could be set in a rectangle way (for example in time, Hz order: [(-0.2, 0), (8., 10.)]) 

You can already do that interactively.

mmagnuski, 2016-11-29:
Oh, I didn't know, thanks! 
Here I meant that the topos in plot_joint would not point to a single point but to a box surrounding the defined tfr region.

jona-sassenhagen, 2016-11-29:
>It may be nice if the channels (set of channels) used for the TFR is highlighted in the topomaps (although the user will always know which channels the TRF is obtained from...).

All. It's a mean over all by default, and I'm not sure other options should be provided.

>Here I meant that the topos in plot_joint would not point to a single point but to a box surrounding the defined tfr region.

The API for topomaps here is already a time point plus window around it parameter, IIRC `times`and `average`. Just make `average` take a (time, freq) tuple and there it is.

The above is actually with 3-Hz freq bins and 200-msec time windows (for the topomaps).

jona-sassenhagen, 2016-11-29:
Is it save to average TF data over multiple channels using plain old `np.mean` btw? I guess they're reasonably normally distributed across channels.

jona-sassenhagen, 2016-12-06:
Any further comments? Don't just wanna go for implementing this if there's little interest. Also I'm sure if you care about this, you'll have opinions on specifics :)

kingjr, 2016-12-06:
Sounds interesting, but I'm not totally clear about the proposal. What is the tfr image showing? the average time frequency over channels? It's std across channels (equivalent of GFP)?

jona-sassenhagen, 2016-12-06:
It's the average. It's done like this in an EEGLAB viz function, used for example in Makeig's Science and PLoS Biology papers.

kingjr, 2016-12-06:
I'm +1. It seems like a helper for mining tfr

agramfort, 2016-12-06:
ok for me

mmagnuski, 2016-12-06:
I like it - but would like an option for it to not be average over all channels (`picks`). So that one could show ROI-based tfrs (or build on this functionality to show cluster-corrected stats etc.).

jona-sassenhagen, 2016-12-06:
w.r.t. @mmagnuski , I think we first need a function to plot the image for a selection of channels, or extend the normal tfr.plot() function like that.

@agramfort , would you be ok with this:

`averageTFR.plot(picks=picks, average=True)`

docstring:

```
average : bool
    If False, show one plot per picked channel.
    If True, show only one plot: the average over all selected channels. If picks
    is empty, the mean over all channels is plotted.
    Defaults to False.
```

And then this one simply adds the topomaps on top.

dengemann, 2016-12-06:
not opposed.

mmagnuski, 2016-12-06:
`average` is good - I previously considered lists of lists so that `[[1, 2], [3, 4, 5]]` produces two figures - one for average of picks 1 and 2 and the other for average of picks 3, 4 and 5. But this is probably less intuitive than `average` param.

PR #4981 (2018-03-29): WIP: Tfr plot joint
Finishing #4134 , which has been sitting in limbo cause @mmagnuski and I are lazy :)

closes #3801

Todo: 
- [x] fix all tests
- [x] docs
- [x] take care of multiple channel types smoothly
closed 2018-03-29 by larsoner