mne-tools/mne-python #3804: [BUG] sources_as_raw leads to error when calling __repr__ of raw object

ppasler, 2016-11-29:
After creating an `ICA object` and calling `get_sources`. The created `Raw object` hast an empty filenames list [mne.preprocessing.ica:_sources_as_raw](https://github.com/mne-tools/mne-python/blob/master/mne/preprocessing/ica.py#L729).
This causes an `IndexError` when calling [repr](https://github.com/mne-tools/mne-python/blob/master/mne/io/base.py#L1884) because the list is not `None`, but empty.

Eric89GXL, 2016-11-29:
Looks like it should set `_filenames = [None]` instead of `_filenames = []`. Can you make a PR to fix it, and add a test? The test can just be a line in `mne/preprocessing/tests/test_ica.py` that does `print(repr(ica_raw))`

dengemann, 2016-11-29:
Wow, this seems like a time-warp, a regression?
Fix welcomed.

ppasler, 2016-11-29:
@Eric89GXL @dengemann sure.

ppasler, 2016-11-29:
@Eric89GXL @dengemann this is first time I contribute to an open source project. 

1. fork the repo
2. create a branch from master
3. add a PullRequest with my changes

Is this correct?

Eric89GXL, 2016-11-29:
Yep that's the idea. How you go about creating branches and the like is up to you. It's possible to do it all graphically on GitHub to start. For frequent development it's easier to learn the `git` command-line interface.

ppasler, 2016-11-29:
I am familiar with git, but not with the workflow for OS projects :)
I'll try it now!

ppasler, 2016-11-29:
It seems not to connect the issue with the PR, do I have to use some specific branch name or commit text?

commit 4ca914d0 (2016-12-02): [MRG+2]: fixed 3804 and added test (#3806)

* fixed 3804 and added test

* Closes #3804; Small format changes.

* removed needless repr

* #3804: added bugfix text

* #3804: reverted unintended change

* #3804: added author to what's new

* #3804: reverted unintended change - again

* #3804: included test to existing test method

* #3804: improved fix description

* #3804: reverted comment improvement

* #3804: fixed failing test

* #3904: removed unnecessary whitespaces

* #3804: changed test expression

* #3804: fixed test
closed 2016-12-02 by Eric89GXL