mne-tools/mne-python #3807: [Feature Request] In-place flag for ICA.apply.

ppasler, 2016-11-29:
As I used "mne.preprocessing.ICA:apply" I noticed, that the method changes the `inst object` in-place. This was confusing to me, as the API says nothing about it and also returns an instance.
I saw it on other places, that there is an `in-place` flag.

I know as a workaround I could copy the instance before applying

dengemann, 2016-11-29:
Yes this the actual way to do it. The inplace flag / copy parameter was a
design flaw. The mne-pythonic way of getting things done is using .copy

Eric89GXL, 2016-11-29:
+1 for better documenting the in-place behavior. I think most places we have an `apply` we do it in-place. I'd rather not add a `copy` param given that you (as you note) can already pass `inst.copy()` and it's just as clear.

ppasler, 2016-11-30:
Better documentation and no return value in the code?

dengemann, 2016-11-30:
> Better documentation and no return value in the code?

Yes. This maybe worth a centrally placed comment in the tutorial, making explicit how to deal with copies and in-place modifications. We could then list which functions actually need to be handled this way and extract some rule for recognising them by the method/function name.

ppasler, 2016-11-30:
For now I would just add it to the apply method?

dengemann, 2016-11-30:
> For now I would just add it to the apply method?

Yes the inline documentation in the code is a start. I'd have a look though how this is currently handled in other places.

Eric89GXL, 2016-11-30:
Typically we still return the instance even during in place operations to
allow chaining, I see no special reason to change that pattern here

ppasler, 2016-11-30:
@Eric89GXL oh I see, that totally makes sense. But the hint in documentation can stay, right?

Eric89GXL, 2016-11-30:
Yeah +1 for improving docs

closed 2016-12-05 by ppasler