mne-tools/mne-python #3811: [ENH] forward model with fixed orientations

Mountainmums, 2016-11-30:
Hi,

are there any plans to add the --fixed functionality for forward modeling to MNE-Python? 

We are trying to migrate our processing pipeline to Python and it would be much simpler to be able to do all our steps within that framework.

Eric89GXL, 2016-11-30:
You can simply use convert_forward_solution after computation.

Mountainmums, 2016-11-30:
Did not know that. Thanks for the tip !

Eric89GXL, 2016-11-30:
Do you have time to make a PR to improve a tutorial? Also in `make_forward_solution` we could say in the `Notes` section that `fixed` is not necessary because you can just `convert_forward_solution`

Mountainmums, 2016-12-01:
Not sure I have the expertise to improve anything, but if I can figure out how to make a PR (GitHub newbie,), I can add to the relevant tutorials and to the make_forward_solution a line about using the convert_forward_solution to get fixed orientations.

PR #3899 (2017-01-12): MRG: Fix docs
Fixes a ton of minor issues with our docs.

Closes #3811.
Closes #3879.
closed 2017-01-12 by agramfort