mne-tools/mne-python #3812: VIZ/FIX: plotting purely negative values in STC object

dengemann, 2016-11-30:
Sometimes the values of an STC object are negative, e.g. when computing the spectral energy in dB for MNE inverse solvers. However, our stc.plot methods do not support such a case, neither by default, nor by `clim`. In such a case the colour would assume uniform values, as if all was zero.
I tell you, this is not good.

cc @Eric89GXL @wronk

wronk, 2016-11-30:
You mean if the values are purely negative?

dengemann, 2016-11-30:
>You mean if the values are purely negative?

yah, absolutely purely totally negative. Got it!

It works via clim if you go by value. But it should work by auto as well and if you go by percentage in clim.

kingjr, 2016-12-01:
(FWIW I think the colorscaling in source space is crazy. The scale is non linear, the colormap doesn't comply to the new standards, and it can be pretty tough your own cmap.)

Eric89GXL, 2016-12-01:
> The scale is non linear

Typically (and also historically) people plot most often some form of statistic that is useful to threshold, and it's often two-sided. So a nonlinear/thresholded cmap like `mne_analyze` makes sense for those use cases.

> the colormap doesn't comply to the new standards

Some of the motivation for using the `mne_analyze` colormap is historical, but the application of plotting on the inflated brain is a bit special. (It's also not *too* terrible as far as divergent colormaps go). Plotting on an inflated brain is a special case in the sense that alpha is important -- it actually produces pretty readable results by going `transparent color 1->opaque color 1->opaque color 2` at the positive and negative ends of the colormap. I don't imagine that a fixed alpha would look as clean. But I'm open to other ideas if we can actually improve on `mne_analyze` default in terms of readability of statistical maps.

> it can be pretty tough your own cmap

This is true. It would actually be pretty easy to support `matplotlib` colormaps -- take the colormap, sample it at 256 even locations, and pass that array to Mayavi, more or less.

kingjr, 2016-12-01:
I agree and know these reasons, but to me, both the non linearity and the threshold invite users to hack their results and hide whatever doesn't go in their favor. 

Alternatives exist: contouring statistically significant regions, avoid thresholding, overlapping the darker sulci in alpha and keep a RdBu_r or YlOrRd cmaps such that the activation colors (saturation) desn't strongly interact with the anatomical info (luminance).

dengemann, 2016-12-01:
I have many usecases where thresholding does not make sense. For example
when you simply show distractions of power or coefficients that are
non-gaussian, or don't have a conceptual 0 point. I then indeed pass the
matplotlib array and avoid transparency, our api then forces me to compute
the fmin, fmid and fmax and pass them explicitly by clim in value mode. And
I think this is still not the same as using that map in matplotlib on topos
for example.
I'm saying that this is not beautiful in terms of usage. And we face the
goal think beyond ERPs and make our tools fun to use on other data.
And it would be nice to have a contours API indeed as JR implies. This is
unrelated though.

Any suggestions how to move here in terms of API design?

wronk, 2016-12-01:
There are definitely plenty of interesting use cases that we actively avoided when we added the ability to define limits with percentages because the API got messy very quickly.

I'm all for more functionality, but I suspect it'll be a pretty big effort and someone will need to champion it. IF we go down that path, it will entail:
* a decision on what use cases we want to handle (sequential maps, custom colormaps, contours, etc.)
* new gallery examples to those use cases (like spectral energy in dB, etc.)
* new documentation or a tutorial explaining all use cases, because I think we're getting close to the point of needing a tutorial just for the recent `clims` functionality
* patience knowing that asking the community what looks "good" will give many diverging (and strong) opinions
* an MNE color champion. Unfortunately, I don't have the bandwidth to volunteer this time around. Maybe a GSoC student?

dengemann, 2016-12-01:
I'd certainly take some of these within the next months. I'm planning some work on documenting some of these cases and tuning the API wherever it makes sense.

dengemann, 2016-12-03:
@wronk @Eric89GXL @kingjr a lot could be improved already by making the call to pysurfer scaling optional, this could be exposed as something like `clim=False`. What do you think?

Eric89GXL, 2016-12-03:
I don't quite get it. Some limits must always be used to map the colormap
limits to some data values, so I'm not sure what you mean. What would that
mode actually do with a given set of data?

Eric89GXL, 2017-02-14:
@dengemann do you have time to make a PR to fix this?

dengemann, 2017-02-15:
No I don't have time now. It will be one of the things I will tackle when
taking a few days to support spectral analysis in source space and some
other things that are on my list.

larsoner, 2019-08-20:
I think we have support for this through one of the recent `clim` refactorings, feel free to reopen if someone hits it again

closed 2019-08-20 by larsoner