mne-tools/mne-python #3814: Partial topomaps (masking?)

mcaffini, 2016-12-01:
In my effort to use MNE Python with **fNIRS** data I find that a major issue is to visualize topomaps. In most fNIRS systems, channels come in limited number and only cover part of the scalp, while in EEG and MEG the whole scalp is usually covered.
It would be usuful to have some sort of **masking support** in plot_topomap, for example using a convex hull approach (quoting @agramfort). I'm not sure whether this can be achieved using other parameters such as **outlines** and **image_mask** and, if so, how.

p.s. Probably a new name for this feature should be introduced, because the parameter name 'mask' is taken (by **mask** and **mask_params**) and is used for a different purpose.

agramfort, 2016-12-01:
could you share a small -ave.fif file to test?

mcaffini, 2016-12-09:
Sorry I had to wait for a vacation day to get back to this.
This is an evoked file with my montage. In the layout plot it does not render properly because it normalizes the coordinates though.

[evoked-ave.zip](https://github.com/mne-tools/mne-python/files/641750/evoked-ave.zip)

agramfort, 2016-12-09:
thanks for sharing

we'll look into it

Eric89GXL, 2017-02-14:
Closing this as a dup of #3987, @mcaffini feel free to subscribe to that issue for updates

closed 2017-02-14 by Eric89GXL