mne-tools/mne-python #3816: [Question] Adding ECG channel to existing EEG raw object?

ppasler, 2016-12-04:
Hi there,

I recorded 14 EEG and 1 ECG channel with different devices. The sample rate for the EEG is roughly 128 and for the ECG 1000Hz. I tried to use `add_channels` with the EEG `raw object` but get an error `'All dimensions except channels must match'`. So I resampled the ECG data to 128Hz, but the data differ in 2 rows.
_So I wondered if there is an automatic way to merge two raw objects?_
_Did I miss some helper method or a tutorial for doing this?_

By the way, I don't get why `add_list` in `add_channels` must be a list, as I am only adding on object.

Thanks in advance,
paul

jaeilepp, 2016-12-06:
> So I resampled the ECG data to 128Hz, but the data differ in 2 rows.

I'm not sure what you mean.

> So I wondered if there is an automatic way to merge two raw objects?
Did I miss some helper method or a tutorial for doing this?
By the way, I don't get why add_list in add_channels must be a list, as I am only adding on object.

Passing a list simply allows adding multiple objects at once. You can always do ``raw.add_channels([raw2])``.

ppasler, 2016-12-06:
@jaeilepp Even with resampling to the same `sfreq` the `_data` array has not the same length (they differ in length of 2), which leads to an error. My "workaround" was the following:

1. Create the EEG `raw object`
2. Create the ECG `raw object`
3. Resample `rawECG` to the `sFreq` of `rawEEG`
4. Crop `rawECG` to `t_max` of `rawEEG`
where `t_max = (eegRaw.n_times - 1) / eegRaw.info['sfreq']`

Assuming EEG and ECG start at the same timestamp.

This works, but seems a bit awkward. If I didn't miss a better option, I'll close the issue for now.

jaeilepp, 2016-12-06:
To me this sounds more like a synchronising issue than an issue with MNE.

ppasler, 2016-12-06:
I'm not sure about that. I have two signals with the same time range but different sample rates, I assumed this was not a big deal - but anyhow I found a solution. I don't know how others record signals with different devices? Do you have any tips how to perform better?

jaeilepp, 2016-12-06:
Yes. When you resample it is not always obvious which samples to include (especially when the sampling frequencies don't align well). For example, see discussions at https://github.com/mne-tools/mne-python/issues/2787 and https://github.com/mne-tools/mne-python/pull/2657

ppasler, 2016-12-06:
I see, to me as noob it was not clear how this works. Resampling and cropping solved it for me, I am not sure how often people do this my way. Maybe it's worth a tutorial - I don't know.

closed 2016-12-06 by ppasler