mne-tools/mne-python #3819: Q: get chan types?

kingjr, 2016-12-06:
Is there a `get_chan_types` somewhere? I'm trying to `create_info` and could use such thing.

Else here's a snippet if someone wants to implement it:

```python
def get_ch_types(info):
    types = {1: 'mag', 301: 'ref_meg', 502:'misc', 3:'stim'}  # to be finised
    ch_types = list()
    for ch in info['chs']:
        ch_types.append(types[ch['kind']])
    return ch_types
```

dengemann, 2016-12-06:
We have something related in the picks module; but also check contains
mixin: "eeg" in raw

mmagnuski, 2016-12-07:
You can use [`channel_type`](https://github.com/mne-tools/mne-python/blob/master/mne/io/pick.py#L17) from mne.io.pick for this:
```python
from mne.io.pick import channel_type
ch_types = [channel_type(info, idx) for idx in range(len(info['chs']))]
```

mmagnuski, 2016-12-07:
Also [channel_indices_by_type](https://github.com/mne-tools/mne-python/blob/master/mne/io/pick.py#L582) may be helpful.
BTW - having these functions in `mne.io.pick` is not super-intuitive

agramfort, 2016-12-07:
@kingjr reopen if it does not do the job for you

closed 2016-12-07 by agramfort