mne-tools/mne-python #3840: Real filter in DICS

brovelli, 2016-12-13:
Hi all,
here is an issue related to #3832
As far as I understand the mne beamforming implementation is the orginal version proposal by Gross et al. (PNAS, 2001). Hipp et al (Neuron, 2011) proposed to derive the filter from only the **_real_** part of the cross-spectral-density matrices instead of the complex cross-spectral density matrix as originally proposed. They argue that [ I cite from their suppl material] "the propagation of the brains’ electrical and magnetical fields can be considered instantaneous (Nunez and Srinivasan, 2006). Thus signals at different EEG/MEG sensors from one source have no phase lag. Consequently, only the real part of the cross-spectral density matrix captures this instantaneous correlation and should be used to derive the filter."

In Fieldtrip there is the possibility to have both complex and real filters using cfg.dics.realfilter='yes'. I would suggest to do the same in mne. I personally use real filters only for the above arguments. To get real filters, , we should simply change for example line 67 in _dics.py.

From:
Cm = data_csd.data
To:
Cm = data_csd.data.real

Or maybe you have implemented this option somewhere else in the code I am missing.
NB: this issue is strongly related to #3832
Thanks

agramfort, 2016-12-13:
I was not aware of this feature in fieldtrip and this Hipp argument.

I would add an option for this in the function.

can you open a PR?

brovelli, 2016-12-13:
Yep, I'll open a PR

PR #3854 (2017-03-19): [MRG+1] Fix reg dics + lcmv
taking over #3839 and #3841 

Closes #3840
Closes #3841
Closes #3832
closed 2017-03-19 by agramfort