mne-tools/mne-python #3846: [Question] How to use `plot_topomap` from raw sensor positions?ensor positions?

nzw0301, 2016-12-14:
I want to plot `Evoked.plot_topomap` without `layout` .
Therefore, it is necessary to set raw sensor positions to `info.chs`.
But  I cannot find a way above.

I have raw sensor positions (#channles, 3) below:

```
id, x,  y,   z
0, 0.8, 0.0, 6.7
...
63, 0.1, 0.0, 6.8
```

thanks.

jaeilepp, 2016-12-14:
You can plot the data with ``mne.viz.plot_topomap``, but you would need to convert the positions to xy-coordinates first.
To populate the info with the positions you would have to do something like:
```
for idx in range(len(pos)):
    evoked.info['chs'][idx]['loc'][:3] = pos[idx][1:]  # first three in loc are xyz
```
I don't know if there's an easier way. I wonder if we should introduce a function for this.

agramfort, 2016-12-14:
you can check the sensor location with:

http://martinos.org/mne/stable/generated/mne.viz.plot_sensors.html#mne.viz.plot_sensors

Eric89GXL, 2016-12-14:
You can create a [Montage](https://mne-tools.github.io/stable/generated/mne.channels.Montage.html#mne.channels.Montage) and then use `evoked.apply_montage(montage)`.

nzw0301, 2016-12-14:
Thanks for replied.
I will do them immediately.

nzw0301, 2016-12-19:
I tried @Eric89GXL 's [comment](https://github.com/mne-tools/mne-python/issues/3846#issuecomment-267052867).

I can create montage.
But `evoked.apply_montage` showed the error message.

`AttributeError: 'EvokedArray' object has no attribute 'apply_montage'`

My mne version is `0.13.1`.

Thanks.

agramfort, 2016-12-19:
try evoked.set_montage

nzw0301, 2016-12-19:
thank @agramfort so much.

closed 2016-12-19 by nzw0301