mne-tools/mne-python #3847: BUG: Plotting TFR with non-linear frequency spacing

nwilming, 2016-12-14:
I've computed a TFR decomposition with logarithmically spaced frequencies. Calling `plot` on the resulting AverageTFR objects shows plots with wrong frequency labels.

Digging a bit deeper I found that plotting is done with a call to matplotlib's imshow function (in `mne.viz.topo._imshow_tfr`)  and then using the `extent` keyword to put appropriate labels onto the plot. However, these labels only work for uniform linearly spaced labels. 

The following example illustrates this point:
```
x, y = logspace(log10(4), log10(150)), logspace(log10(4), log10(150))
Z = dot(x[:, newaxis],(y[newaxis,:]*0)+1)
extent = (x.min(), x.max(), y.min(), y.max())
subplot(1,2,1)
imshow(Z, origin='lower', cmap='gray', extent=extent, aspect='auto')
title('imshow')
subplot(1,2,2)
pcolormesh(x, y, Z, cmap='gray')
ylim([y.min(), y.max()])
xlim([x.min(), x.max()])
title('pcolor')
```
[image]

`pcolor` correctly displays the spacing between points on the x and y-axis, but imshow does not. The imshow plot gives the impression that values for 80 on the y-axis are low (black), while they actually are quite high (gray). 

One way to fix this would be to use `pcolormesh` instead of `imshow` for plotting. The only tricky thing would be to get the edges right that are needed for `pcolormesh`. I have a test implementation of a `_pcolormesh_tfr` that I could share. But I suspect fixing this behavior would need changes to a all of the plotting functions in AverageTFR - insofar I'd be interested in your opinion on how to best fix this issue.

mmagnuski, 2016-12-14:
I had this issue before. The problem is that theoretically you can have arbitrary spacing in your freqs of interest. Simplest solution would be to allow log y axis (or detect this automatically). Another simple solution is to just change the y tick labels (this is what I have been doing so far). The y extent is still linear then however and adding things onto the axis (bounding boxes highlighting ROI for example) might give unexpected results.

agramfort, 2016-12-14:
indeed this needs a fix.

PR or at least a gist welcome

thx

Eric89GXL, 2016-12-14:
> The problem is that theoretically you can have arbitrary spacing in your freqs of interest. 

That isn't really a problem if we use a `pcolor` variant like `pcolormesh` because it will take that spacing into account.

Yes please open a PR for the `pcolormesh` version if you're up for it

mmagnuski, 2016-12-14:
@Eric89GXL ok, I actually never used `pcolormesh`. Can you still have interpolation then?

Eric89GXL, 2016-12-14:
You can do `shading='gouraud'` which might be the same

Eric89GXL, 2017-01-11:
Actually according to the `mpl` folks there is a `NonUniformImage` class we can use.

@mmagnuski do you have time to try it in the next couple of days? It would be good to get this in for 0.14. If not, I can probably do it.

mmagnuski, 2017-01-11:
Yes, I can work on this this week. I looked into the mpl example of NonUniformImage and it is perfect for this indeed.

PR #3905 (2017-01-23): MRG+1: nonuniform image option in tfr plot
fixes #3847
This is not finished API-wise. The way it works currently is that there is an `imtype` arg (temporary name - any ideas?) that allows `'normal'` or `'nonuniform'` and thus changes the plotting artist to be NonUniformImage.
I make it dependent on a kwarg because:
* NonUniformImage is a little less flexible - does have only two interpolation options for example ('none' and 'bilinear')
* I did not check it but it may be slower
* having a kwarg opens the door to adding `'logy'` option (next PR maybe - if you think that would be useful, see the  [... cut]
closed 2017-01-23 by Eric89GXL