mne-tools/mne-python #3850: the 'spatial_exclude' parameter of the function 'spatial_temporal_clustering'

dongqunxi, 2016-12-15:
Dear all,

I want to get the significant clusters on the cortices, the function ```spatial_temporal_clustering``` estimates the vertices on the surface of the whole brain. I do not want to take the medial central vertices into account. since they do not include any known cortical regions. In my understanding of the parameter  'spatial_exclude', it can exclude the vertices of this region. 

My question is that if it makes sense, how do we get the indices of these vertices to be excluded?

Thanks a lot!

Eric89GXL, 2016-12-16:
There are `lh.Medial_wall.label` and `rh.Medial_wall.label` in `fsaverage/label` that comes with FreeSurfer, though see also 

https://mail.nmr.mgh.harvard.edu/pipermail//freesurfer/2013-July/031908.html

closed 2016-12-16 by Eric89GXL

dongqunxi, 2016-12-16:
Thanks a lot, Eric. One additional question, some of the indices from
lh.Medial and rh.Medial regions are the same, if I set the parameter
'spatial_exclude', how to organize the list of indices for excluding?

Best wishes,
Qunxi Dong

Eric89GXL, 2016-12-16:
Are you using the standard `fsaverage` surface (ico-5)? If so then you need to add `10242` to the right-hemisphere vertex numbers.

If you are using a different subject or different surface, it's more complicated because you have to deal with indexing indices.

dongqunxi, 2016-12-16:
Thanks a lot for your help, I am using (ico-5). :-)

Best wishes,
Qunxi Dong