mne-tools/mne-python #3853: normalize lcmv / dics solution

agramfort, 2016-12-15:
from the mailing list and for the record:

https://mail.nmr.mgh.harvard.edu/pipermail//mne_analysis/2016-December/003801.html

```
in fact, I did not look into the details of the lcvm code. For what concerns dics, I think the output is the ratio of powers (data/noise), that is what is normally called "neural activity index" NAI. In don't think the 1997 Van Veen paper has defined the NAI (https://www.lsv.uni-saarland.de/fileadmin/teaching/dsp/ss15/BF_VeenBuckley.pdf). For a ref about NAI, you can look at this chapter:

http://www.ccs.fau.edu/~fuchs/pub/Beamer.pdf

Or the webpage of Fieldtrip:

http://www.fieldtriptoolbox.org/tutorial/beamformer#neural_activity_index

More generally, it would be good to have the possibility in MNE to have different normalization techniques (in addition to "None", in some case people may prefer not to normalise wrt noise). Keeping in mind that power values are not normally distributed (they are chi-squared), some ways to normalise are:

1) dB = 10 * log10(data/noise)

2) z-score = [data - mean(noise)] / std(noise)

3) z-score of logs = [log(data) - mean(log(noise))] / std(log(noise))  (the log-transform make power values approx. gaussian, as sqrt-transform)
4) ERD/S = [data - mean(noise)] / mean(noise)

... just to cite a few
Some inspiration here (lines 198-207 ):

https://github.com/fieldtrip/fieldtrip/blob/master/ft_freqbaseline.m
```

Eric89GXL, 2017-03-21:
@agramfort do we still need these (e.g., normalization options), or did your PR sufficiently cover it?

agramfort, 2017-03-21:
yes nothing is done here. Will be done this summer I guess.

larsoner, 2018-04-04:
@britta-wstnr can you comment on the progress / relevance here?

britta-wstnr, 2018-04-04:
Neural activity index is done (``weight_norm="nai"``).

I never looked at high SNR breakages, but in general they are a common and inherent problem with spatial filters.

The normalizations are not implemented for beamformers AFAIK. Might be an interesting extra function (to apply to already computed output?) and is probably most relevant if returning power - the source power LCMV implementation ``lcmv_source_power`` needs some polishing at some point I think (it's not been touched last summer), so it might be logical to tackle it then - if still relevant?

larsoner, 2019-03-26:
I can bring some high-SNR phantom data to the sprint so we can interactively break things there.

britta-wstnr, 2019-04-25:
Do we want to look more into this? What would be the agenda of this (I mean, after the fun of breaking things - the breakage is inherent to the design of spatial filters ...)?

larsoner, 2019-04-25:
If it's inherent, there is no known fix, and we should just learn to live with it, we should close

closed 2019-04-25 by larsoner