mne-tools/mne-python #3855: ica.info["filename"] ?

jona-sassenhagen, 2016-12-15:
If I understand it correctly, `ica.info["filename"]` can be
- the name of the ica-fif - file on disk, if the ICA object was read from disk
- the name of the data file used for fitting

This is 1. a bit confusing, 2. potentially dangerous if you use something like

`ica.save(ica.info["filename"])`

See what I mean?

I think it should either 1. always default to the file used for fitting, or 2. to None if not read from disk and to the file name if read from disk.

agramfort, 2016-12-16:
indeed

the filename is not part of the meas info structure. Only the fif reader adds it (I think).

I think we should remove it.

@Eric89GXL WDYT?

Eric89GXL, 2016-12-16:
It's not part of the measurement info, right. Actually I don't even think the FIF raw reader should add it. That seems like a bug to me. I'm +1 for removing it. We actually have the filename(s) stored elsewhere in `BaseRaw` already.

Eric89GXL, 2016-12-16:
I'll fix it

closed 2016-12-21 by agramfort

ppasler, 2017-01-10:
Why throwing an exception? Wouldn't a warning be enough? It breaks my `Raw` object creation routine.

agramfort, 2017-01-10:
can you be more explicit?

ppasler, 2017-01-10:
I used the filename attribute in some places in my application, e.g. to differ objects or plots. 
It seems plausible to remove the attribute from the API, but I don't think it's necessary to throw a [KeyError](https://github.com/mne-tools/mne-python/blob/master/mne/io/meas_info.py#L258).

I think a deprecation warning for 0.14 and maybe an error om 0.15 would be better.

Eric89GXL, 2017-01-10:
We can get rid of the `KeyError`. IIRC it was there mostly to ensure internally that we were not setting it. Can you make a PR to remove the `KeyError`? A warning or getting rid of the check altogether, either way is fine with me.

ppasler, 2017-01-10:
Alright, I will propose a solution.

PR #3860 (2016-12-21): MRG: Remove misleading info["filename"]
Closes #3855.