mne-tools/mne-python #3859: [Question] Does the topomap use the ica unmixing or mixing matrix?

ppasler, 2016-12-16:
Hey there,

I just debugged the `ICA.plot_components()` method and stumbled over the docstring in [topomap.py#L778](https://github.com/mne-tools/mne-python/blob/master/mne/viz/topomap.py#L778), where it says:

> Project unmixing matrix on interpolated sensor topogrpahy.

But here the mixing matrix is used: [topomap.py#L889](https://github.com/mne-tools/mne-python/blob/master/mne/viz/topomap.py#L889).

Maybe this is some language or math understanding problem of mine.
Could somebody clear this up and shortly explain how the topomap creation works inside?

Thanks in advance,
paul

agramfort, 2016-12-16:
what you should plot as the components are the columns of the mixing matrix

ppasler, 2016-12-16:
So it's actually the mixing matrix and not the UNmixing matrix? 
The map shows the scaling factors for one component (column in the mixing matrix) based on the topography of the sensors?

When I look into the code, there is much more done. I am just curious what else is done to improve the plots.

jona-sassenhagen, 2016-12-16:
Hm ... https://github.com/mne-tools/mne-python/blob/master/mne/viz/topomap.py#L889 should probably use `ica.get_components` now, for consistency's sake.

ppasler, 2016-12-16:
@jona-sassenhagen doesn't the term "component" in case of the ICA refer to the source signals (= independant components)? I am slightly confused :)

mmagnuski, 2016-12-16:
@ppasler - mixing matrix is what you want to plot because it tells you how each component projects to sensors (so it is "normalized" with respect to components signal). When you plot the UNmixing matrix on the scalp you'll see that it is more difficult to understand - different channels have different levels of signal and noise and both are important in creating a spatial filter that separates relevant signal (for a full explanation see the discussion on [filters vs patterns](http://ac.els-cdn.com/S1053811913010914/1-s2.0-S1053811913010914-main.pdf?_tid=5540c92e-c3ad-11e6-9fd1-00000aacb35f&acdnat=1481906170_9932edcc38207273af36881233166209)). A short example of filters vs patters from [a related paper](http://citeseerx.ist.psu.edu/viewdoc/download?doi=10.1.1.723.9847&rep=rep1&type=pdf):
[image]

For the term "component" - it is true that it may refer to source signal - however the name `get_components` was chosen to be consistent with `ica.plot_components` (plots topographies of mixing matrix) vs `ica.plot_sources` (plots source signal).

ppasler, 2016-12-28:
@mmagnuski thanks for the papers, they clear things up. So the mixing matrix is taken for plotting.

closed 2016-12-28 by ppasler