mne-tools/mne-python #3870: mawxell_filter with CTF data

romquentin, 2016-12-21:
Hi All, 
When running 
```
raw_sss = maxwell_filter(raw, origin='auto', st_duration=10.0,
                                     coord_frame='head',
                                     destination=raw_dest,
                                     ignore_ref=True, verbose=True)
```
I got the following error: 
`ValueError: trans must go meg<->head, provided trans is ctf_head->head
`
I checked that I have a `dev_head_t` in `raw_dest.info`. 
I have the same issue before having set a montage or after.

Thank you!

Romain

Eric89GXL, 2016-12-21:
What is `raw.info['dev_head_t']`?

Can you share a problematic file somewhere?

romquentin, 2016-12-21:
@Eric89GXL Please see the link to the data. 
https://www.dropbox.com/sh/q73kkw9tllavhgx/AABoiZmkYl3RrQoSGcgp44W2a?dl=0
The` raw_dest` is the destination location for the head. 
The info['dev_head_t'] is: 
`<Transform  |  MEG device->head>
[[  9.99036014e-01   2.97366362e-02   3.22922021e-02   1.94156461e-03]
 [ -3.16679031e-02   9.97632802e-01   6.10405616e-02   2.34328065e-04]
 [ -3.04006208e-02  -6.20043464e-02   9.97612774e-01   7.58210793e-02]
 [  0.00000000e+00   0.00000000e+00   0.00000000e+00   1.00000000e+00]]
`
To get the error described above:

`from mne.io import Raw`
`from mne.preprocessing import maxwell_filter`

`raw = Raw('/Users/quentinra/Dropbox/test_maxwellfilter/raw.fif', preload=True)`
`raw_dest = '/Users/quentinra/Dropbox/test_maxwellfilter/raw_dest.fif'`

`raw.apply_gradient_compensation(0)`
`raw_sss = maxwell_filter(raw, origin='auto', st_duration=10.0,
                                     coord_frame='head',
                                     destination=raw_dest,
                                     ignore_ref=True, verbose=True)`

Thanks a lot!

Eric89GXL, 2016-12-21:
Oh you're passing a `Raw` filename to `trans`. It just calls `read_transform` and the first transform it gets is probably a CTF transform. Can you try adding this line?
```
raw_dest = mne.io.read_raw_fif(raw_dest).info['dev_head_t']
```
I guess we could make `read_trans` (which is used under the hood) smarter to not just take the first transform found.

romquentin, 2016-12-21:
oh, ok! Thanks. It works until this error: 
`Only 0 head digitization points of the specified kinds ("eeg", "extra",), at least 4 required`
Do I need a headshape to use maxwell_filter?

Eric89GXL, 2016-12-21:
With `origin='auto'` you do, but you can always set it to something reasonable depending on your subject age like `(0., 0., 0.04)` for adults

romquentin, 2016-12-21:
Got it. Thank you!

closed 2016-12-21 by romquentin