mne-tools/mne-python #3871: When using mne.viz.iter_topography, on_pick callback is called with pyplot instance instead of the axes

Teekuningas, 2016-12-21:
This is the line:

https://github.com/mne-tools/mne-python/blob/master/mne/viz/topo.py#L223

Shouldn't it be the axes? This would suggest so: http://martinos.org/mne/stable/auto_examples/visualization/plot_topo_customized.html

Thanks,
Tk

agramfort, 2016-12-21:
what fix do you suggest?

Teekuningas, 2016-12-21:
As even the docstring of the iter_topography says: 

"on_pick : callable | None
        The callback function to be invoked on clicking one
        of the axes. Is supposed to instantiate the following
        API: `function(axis, channel_index)`",

I would be as daring as replacing the line "show_func(plt, ch_idx)" with "show_func(ax, ch_idx)" and let people fix their code (they can always introduce pyplot themselves if they want minimal changes)..

-- Tk

Eric89GXL, 2016-12-21:
+1 that sounds like a bug in our implementation

Are you up for making a PR?

Teekuningas, 2016-12-21:
I can do that, yes

-- Tk

PR #3873 (2017-01-09): MRG+1: Fix _plot_topo_onpick (#3871)
Should fix #3871. I went through all mne-python's own related callback functions and seems they can all handle both axes object or pyplot coming.
closed 2017-01-09 by jaeilepp