mne-tools/mne-python #3876: EDF picks on load due to different sfreq

agramfort, 2016-12-22:
when loading an EDF that contains EEG at 256Hz and ECG at 512Hz all my channels get resampled to 512Hz (expected) but the API does not allow me to restricted the read to EEG only, and therefore the resampling which is slow.

any hint? suggestion?

agramfort, 2016-12-22:
cc @Slasnista

Eric89GXL, 2016-12-22:
I'm not sure of good options.

It can't be automatic, it would be bad to have `raw[eeg_picks]` give one sample rate and `raw[:]` give another.

I guess we could add something to the class constructor that allowed you to choose a sample rate, or a different data-fetching method with options for output sample rate (or that did output the lowest-possible one based on `picks`).

agramfort, 2016-12-22:
I see two options:

- add an include / exclude mechanism in the constructor
- allow to drop_channels even if preload=False and then do load_data.

I think 1 is easier...

closed 2017-01-17 by agramfort