mne-tools/mne-python #3878: `mne.viz.plot_topomap()` for MEG data and extracting the `pos` parameter.

DominiqueMakowski, 2017-01-02:
Hey,

I have an array of MEG data corresponding to averaged values for 306 channels (MAG: 102, GRAD: 204). I'd like to visualize a topographic plot of this map. However, the `mne.viz.plot_topomap()` requires a `pos` argument that I have no idea how to extract from the `raw` object.

Moreover, as my data has two channel types (GRAD and MAG), it won't infer it from the raw.info object. 

How can I extract the `pos` parameter or make my way around the `Multiple channel types` problem?

Thanks!!

jona-sassenhagen, 2017-01-02:
Pick the info object with pick_types? The help should say that IIRC

DominiqueMakowski, 2017-01-02:
but it seems that `pick_types()` only allows for a global `meg` parameter that picks both GRAD et MAG channel types, thus making it impossible to plot using `mne.viz.plot_topomap()` 😢

agramfort, 2017-01-02:
@DominiqueMakowski make an EvokedArray using the info from the sample dataset and then use the evoked.plot method.

DominiqueMakowski, 2017-01-02:
It works!

Here's the code:

```python
values  # (306, ) numpy array
values = np.array(values, ndmin=2).T  # Transpose it to a 2D array
evoked = mne.EvokedArray(values, raw.info, tmin=0)  # Convert it to a EvokedArray
evoked.plot_topomap(times=0)
```

Thanks 😄

closed 2017-01-02 by DominiqueMakowski