mne-tools/mne-python #3880: Compute_covariance NaN values

WilliamCollishaw, 2017-01-03:
Ubuntu 16.04 
MNE 0.14.git

`Traceback (most recent call last):`
  `File "pythonScripts/pipeline.py", line 359, in <module>`
    `main()`
  `File "pythonScripts/pipeline.py", line 289, in main`
    `noiseCov = compute_covariance(epochsList, tmin=None, tmax=0, ``method='auto',return_estimators=True, verbose=True, n_jobs=1,projs=None)
  File "<string>", line 2, in compute_covariance`
 ` File "/usr/local/lib/python2.7/dist-packages/mne/utils.py", line 707, in verbose
    return function(*args, **kwargs)`
 ` File "/usr/local/lib/python2.7/dist-packages/mne/cov.py", line 796, in compute_covariance
    scalings=scalings)`
 ` File "/usr/local/lib/python2.7/dist-packages/mne/cov.py", line 928, in _compute_covariance_auto
    stop_early=stop_early)`
 ` File "/usr/local/lib/python2.7/dist-packages/mne/cov.py", line 1036, in _auto_low_rank_model raise RuntimeError('Oh no! Could not estimate covariance because all '`
`RuntimeError: Oh no! Could not estimate covariance because all scores were NaN. Please contact the MNE-Python developers.`

Eric89GXL, 2017-01-04:
Can you upload the `epochs` instance somewhere? It looks like the `compute_covariance` call is given above.

Also, `np.isfinite(epochs.get_data()).all() is True`?

Eric89GXL, 2017-02-14:
Closing since I haven't heard back, but @WilliamCollishaw if you can get the data somewhere we can reopen

closed 2017-02-14 by Eric89GXL