mne-tools/mne-python #3883: find_bads_eog when data is not preloaded and channels differ: bad error msg

jona-sassenhagen, 2017-01-04:
When you use `ica.find_bads_eog(inst)` on an instance that is not preloaded, you may get

```
---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
<ipython-input-39-e371fefd815e> in <module>()
----> 1 ica.find_bads_eog(epochs)

/home/jona/tools/mne-python/mne/preprocessing/ica.py in find_bads_eog(self, inst, ch_name, threshold, start, stop, l_freq, h_freq, verbose)

/home/jona/tools/mne-python/mne/utils.py in verbose(function, *args, **kwargs)
    706         with use_log_level(verbose_level):
    707             return function(*args, **kwargs)
--> 708     return function(*args, **kwargs)
    709 
    710 

/home/jona/tools/mne-python/mne/preprocessing/ica.py in find_bads_eog(self, inst, ch_name, threshold, start, stop, l_freq, h_freq, verbose)
   1095 
   1096         if inst.ch_names != self.ch_names:
-> 1097             inst = inst.copy().pick_channels(self.ch_names)
   1098 
   1099         if not hasattr(self, 'labels_') or self.labels_ is None:

/home/jona/tools/mne-python/mne/channels/channels.py in pick_channels(self, ch_names)
    627         _check_excludes_includes(ch_names)
    628         idx = [self.ch_names.index(c) for c in ch_names if c in self.ch_names]
--> 629         self._pick_drop_channels(idx)
    630         return self
    631 

/home/jona/tools/mne-python/mne/channels/channels.py in _pick_drop_channels(self, idx)
    682         if isinstance(self, (BaseRaw, BaseEpochs)):
    683             if not self.preload:
--> 684                 raise RuntimeError('If Raw or Epochs, data must be preloaded '
    685                                    'to drop or pick channels')
    686 

RuntimeError: If Raw or Epochs, data must be preloaded to drop or pick channels
```

I think in this case because the ICA object has different channels than the instance.

The error message is not helpful I think. Alternatively, the problem should be circumvented somehow altogether.
@dengemann ?

agramfort, 2017-01-05:
yes maybe this can be avoided as it can create big memory allocations

dengemann, 2017-01-22:
Sorry for coming late, adding an explicit @dengemann gmail filter turned
out to be a good idea. I think, in the meantime we have already started to
discuss a pull request in an issue.

closed 2017-01-23 by dengemann