mne-tools/mne-python #3889: ica.get_sources has strange ch_names and ch_type

ppasler, 2017-01-09:
After calling `ica.get_source()` with a `Raw` object I noticed that

* the `ch_names` "ICA 001" start by 1 instead of 0 (which is not consistent)
* I couldn't plot it right, as the `ch_type` for scaling is not clear (it seems to be `misc`, but I couldn't find any documentation about it)

I would suggest to let the ICAs start by 0 everywhere.
Remove the `+1` here: https://github.com/mne-tools/mne-python/blob/master/mne/preprocessing/ica.py#L798

agramfort, 2017-01-09:
this question has been raised so many times... So far we agreed that when you read "ICA #1" it means index 1 but then names start with 1 to be consistent with standard EEG channels on Elekta : EEG001 etc.

do we really have to change this?

ppasler, 2017-01-10:
Sorry, I did not check for older tickets.
You've got a point with consistency of EEG channel naming. 

I just noticed, that inside the `ICA` object the ICs start with 0 (e.g. `ica.plot_sources`), but after transferring to a `Raw` object it starts with 1 - so I would propose to change this within the `ICA` object. It's a common computer science problem that indices start with 0...

The other thing is to add the hint 'the channel type is misc' when calling `ica.get_source` to the doc.

agramfort, 2017-01-10:
channel names start with one and indices eg in  exclude start at zero. Also
ica plot methods refer to ICA indices and start at zero.

+1 to clarify the doc

jasmainak, 2017-01-10:
It seems to be a good candidate for the FAQ section :)

mmagnuski, 2017-01-10:
I also think that ic numbering inconsistency is counterintuitive (and changing the docs won't change the intuitions ;) ). 
`ica.plot_sources` plots sources with names in `IC #000` convention - which is good as it makes comparisons between `plot_components` and `plot_sources` easier, but further increases surprise when `ica.get_sources().plot()` gives other labels (`ICA 001`). If channel labels must of form `ICA 001` - then maybe ICA plotting functions should print `ICA 001` instead of the index? Or maybe both should change to `IC 001`? 😈

Eric89GXL, 2017-06-15:
So over in #4320 the crushing weight of a half dozen (or more) opened issues/PRs about this issue has caused @agramfort to have a change of heart to accept the possibility of naming channels as `ICA 000`+. It creates inconsistency with other channel names, but IMO (and maybe @agramfort has come around to it, now, too) this is less bad / create fewer problems than our current solution, where most users are confused by our ICA interface.

@dengemann @jona-sassenhagen @jasmainak @mmagnuski are you okay with this change?

@mmagnuski would you perhaps be up for making a PR?

jona-sassenhagen, 2017-06-15:
I'm in favour.

dengemann, 2017-06-15:
I'll let you converge on this. After 10 MNE workshops I learned that user
logic is sometimes as important as designer logic :) let's make it trivial
and unsurprising to use ICA!

mmagnuski, 2017-06-15:
I'm also +1 on this. I'm travelling now, but I'll try to get to it this week and make a PR.
(BTW - is anyone attending cutting eeg?)

Eric89GXL, 2017-06-15:
Actually @mmagnuski I had a look and the [past](https://github.com/mne-tools/mne-python/pull/3633) [three](https://github.com/mne-tools/mne-python/pull/2019) [attempts](https://github.com/mne-tools/mne-python/pull/1622) all do the same thing as the [current attempt](https://github.com/mne-tools/mne-python/pull/4320), except #1622 which had the option on how to name them, which we probably need. I'll see if it can be worked into #4320 because the repo for #1622 is gone.

dengemann, 2017-06-15:
Maybe we can also make explicit in the label that it's an index.

teonbrooks, 2017-06-16:
to add to the list of past issues: https://github.com/mne-tools/mne-python/issues/2000 ;)

Probably said already but one of the most confusing parts has been the `IC` vs`ICA` labelling. how do we deal with indexing channels because this problem doesn't come up? Two options:
- get rid of `IC #` completely and remain consistent in `ICA` label and use the ch indexing methods we use for other channel types.
- index with 0 in the name. used to be my preferred option but not such a strong advocate for/against it anymore

larsoner, 2019-08-21:
I think this has been fixed in `master` via naming and/or doc updates

closed 2019-08-21 by larsoner