mne-tools/mne-python #3890: How to remove duplicate channels?

madhavpr, 2017-01-10:
Hi everyone, 

I'm a first time user of the MNE library. I'm trying to read a bunch of EDF files (from the CHBMIT EEG database). I type the following code on my Python interpreter and get this error.

filename = 'chb04_07.edf'
raw = mne.io.read_raw_edf(filename)

RuntimeError: Channel names are not unique, found duplicates for: set([u'T8-P8'])

[Note: As many as 100 files or so have caused this problem]

I understand one channel (T8-P8) has been duplicated but I don't know of a straightforward way to remove that entry. Is there any way to get out of this problem? I really need to read them all. 

Thanks a lot
- Madhav

jaeilepp, 2017-01-10:
I'm not sure if we want to change the current behavior since having duplicate channels can cause problems later. My advice would be to try to fix the files by commenting out the part that raises the error and then
1. read the file.
2. drop the duplicate channel,
3. save it as a fif-file.

After that make sure to uncomment the line that raises the error so it works like before. I checked the file and this work flow seems to work fine.
```
import mne
raw = mne.io.read_raw_edf('chb04_07.edf', preload=True)
raw.drop_channels(['T8-P8'])
raw.save('chb04_07_raw.fif')
```

jaeilepp, 2017-01-10:
Actually, seems like this is somewhat related to https://github.com/mne-tools/mne-python/issues/3876. So perhaps we shall introduce an exclude parameter to help deal with stuff like this.

agramfort, 2017-01-10:
it's related to this:

https://github.com/mne-tools/mne-python/pull/3813

I think it's fixed in master?

jaeilepp, 2017-01-10:
> I think it's fixed in master?

No, that was for the tal channel combination. This data file here actually has one channel doubled. Seems like some kind of error in the acquisition.

agramfort, 2017-01-10:
ok then I would suggest to deduplicate at reading time by appending
some -1 or -2 to the channel name. An raise a warning.

closed 2017-01-17 by agramfort