mne-tools/mne-python #3891: Run corrmap with different amount of ICs

ppasler, 2017-01-10:
I have several datasets with 14 channels ( ~ 14 ICs), for one proband there was an error with one channel. So I had to mark it as bad and only calculate 13 ICs.

When running `corrmap()` I get this error:

	 File "E:\current-adas\project\code\src\util\mne_util.py", line 147, in labelArtefact
		return corrmap(icas, template=template, threshold=0.85, label=label, show=False, ch_type='eeg', verbose=True)
	  File "<string>", line 2, in corrmap
	  File "C:\Python27\lib\site-packages\mne-0.14.dev0-py2.7.egg\mne\utils.py", line 708, in verbose
		return function(*args, **kwargs)
	  File "C:\Python27\lib\site-packages\mne-0.14.dev0-py2.7.egg\mne\preprocessing\ica.py", line 2389, in corrmap
		nt, mt, s, mx = _find_max_corrs(all_maps, target, threshold)
	  File "C:\Python27\lib\site-packages\mne-0.14.dev0-py2.7.egg\mne\preprocessing\ica.py", line 2162, in _find_max_corrs
		all_corrs = [compute_corr(target, subj.T) for subj in all_maps]
	  File "C:\Python27\lib\site-packages\mne-0.14.dev0-py2.7.egg\mne\utils.py", line 2382, in compute_corr
		return (fast_dot(X.T, Y) / float(len(X) - 1)) / (x_sd * y_sd)
	ValueError: shapes (14,) and (13,13) not aligned: 14 (dim 0) != 13 (dim 0)

I wonder if this is normal and how I could fix this?

A workaround could be to add a dummy IC (with 0) to have 14 or calculate only 13 for the other probands. Which both seems not to be stable.

jona-sassenhagen, 2017-01-10:
Isn't the problem with the channel number rather than the component number?

I forgot if the original corrmap interpolates the maps, but this version doesn't (for now).

ppasler, 2017-01-10:
I thought the amount of components shouldn't be greater than the amount of channels, so it depends on each other.

I am running 0.14dev, is there any fix in the pipeline?

jona-sassenhagen, 2017-01-10:
>I thought the amount of components shouldn't be greater than the amount of channels, so it depends on each other.

No, but they can be fewer - you an estimate 13 components for each subject.

Really, I think the problem is with channel numbers, not component numbers.

>I am running 0.14dev, is there any fix in the pipeline?

I'm not planning on working on interpolation any time soon. You can interpolate your channels before fitting ICA.

ppasler, 2017-01-10:
Thanks for your thoughts.

So you suggesting to estimate 13 components for every proband. 
Isn't there any other way? Assume I'll get another proband and only 12 channels are good, then I have to run the ICA for all other probands with 12 components again.
Couldn't I fill the missing component with zeros? Would this have an impact on the result?

Sure the problem are the channel numbers, but they affect the max. number of components.

How would I interpolate my channels?

jona-sassenhagen, 2017-01-10:
>Couldn't I fill the missing component with zeros? 

This would make the data rank deficient and mean you could also only estimate fewer components. Same with interpolation actually.

Personally, I just round down to the nearest nice number: if I have 64 channels, I do 60 components.
There is also an option for selecting enough components to get x% variance explained.

You can interpolate channels with an instance method, I think it's Epochs.interpolate_bads.

ppasler, 2017-01-10:
> Personally, I just round down to the nearest nice number: if I have 64 channels, I do 60 components.
Alright, maybe I'll just try with less components. I thought the more components the better :)
With 14 channels, how many components would you use. My major goal right now, is to extract EOG artifacts.

> There is also an option for selecting enough components to get x% variance explained.
I don't get it, what do you mean?

Do you know if `interpolate_bads` works for channels that are zero all the time (there was a bug within the recording software).

jona-sassenhagen, 2017-01-10:
I'm pretty sure it completely ignores the channels to be interpolated.

ppasler, 2017-01-11:
@jona-sassenhagen you were right, the number of channels is the problem - now I know what you meant, during the last 2 posts. Reducing the component number does not solve the issue, as the shape still differs.

jona-sassenhagen, 2017-01-27:
Can we close this issue @ppasler ?

sappelhoff, 2018-03-29:
Unfortunately I am currently running into the same problem. I want to run Corrmap ... but to get good ICA decompositions I want to reject a channel here and there throughout the dataset so that I ent up with participants having between 60 and 64 channels with different electrodes missing. Thus, the IC maps are also different and corrmap will not be able to do proprt correlations.

From the [Viola et al. 2009](https://www.ncbi.nlm.nih.gov/pubmed/19345611) paper alone, it is not clear how the original `corrmap` deals with differing numbers of channels across data sets:

> The default ICA algorithm used by EEGLAB (Delorme and Makeig, 2004) is Infomax ICA, where the number of ICs is usually equal to the number of EEG channels, normally corresponding to the rank of the data. However, CORRMAP also accepts a different number of ICs per dataset (in case of rank-deficiency or prior dimensionality reduction), thus providing greater flexibility. CORRMAP can also deal with variations in EEG channel numbers within a dataset, such as happens in the case of defective channels. In this case, CORRMAP requests a channel configuration file, and the inverse weights for missing electrodes are then automatically replaced using a modified version of the EEGLAB function eeg_interp().

However, if they interpolate the EEG channels before doing the ICA, that would mean that they do ICA on rank-deficient data ... so they must have a different fix ... maybe someone can find out by looking at [the source (zip file)](http://www.debener.de/corrmap/corrmap1.03.zip).

For now, the only solution I see is keeping the bad channels and doing my ICA decomposition with the bad data, rejecting only particularly bad segments. This suffices for the components I am interested in (blink, horizontal eye movement, ecg) ... but it's not optimal.

larsoner, 2018-03-29:
> However, if they interpolate the EEG channels before doing the ICA, that would mean that they do ICA on rank-deficient data

This is a case we need to support, both for EEG reference (which reduces rank) and MEG processing such as `maxwell_filter` (which reduces from 306 to 80). So we should support it properly, documenting any necessary caveats.

Then you can maybe use interpolation -- though we'd need to check that downstream processes such as source estimation don't mis-estimate the rank.

sappelhoff, 2018-03-30:
@jona-sassenhagen, by "estimating fewer components" do you mean performing a dimension reduction via PCA before the ICA? Or what else do you mean?

In case of the PCA reduction, I see how this applies to ICA and rank deficient data in general, but how would it help for the case of `corrmap`? I thought the issue was that `corrmap` is correlating the "scalp maps" of the ICs and that hence, the mapping between an IC and its corresponding "electrode position" is important. Imagine the case where we have two participants A and B:

- Participant A has electrodes Cz, Pz, ..., Oz (64 electrodes)
- Participant B has the same electrodes, however Cz was bad so it is removed (63 electrodes)

We do an ICA on the pca-reduced data (let's say, 50 components). So although the dimensions of the unmixing matrices would be comparable ( shape 50,50), the IC "scalp maps" that are used as input in `corrmap` are not comparable (shape 64,64 vs. shape 63,63).

Or is `corrmap` doing the correlations on the unmixing matrices instead of the scalp maps? That would surprise me, because the intuition is that scalp maps (e.g., for eye blinks) are stable because they get their physiological grounding from the mapping between electrode position and IC weight ... whereas if the correlation is done on unmixing matrices that have been PCA-reduced before, each row of the unmixing matrix does no longer map to an electrode position (but rather, to a ranking in terms of variance explained), and if we then correlate across participants, an eye blink component for example could have completely different topographies.

sappelhoff, 2018-08-06:
@jona-sassenhagen the questions is still unresolved. In my work, I decided to not perform any dimension reduction and I did not reject channels prior to using corrmap ... thus I had a workaround for this problem.

The API for ICA/PCA is still kind of confusing and not perfectly well documented. In conjunction with using corrmap, this is additionally hard to wrap the head around ;-)

AmirDJV, 2019-03-06:
Is there any development regarding this issue? as unfortunately I'm running into the same problem.

jona-sassenhagen, 2019-03-06:
The issue is you have rejected channels, so your IC maps have different dimensionalities between subjects?

AmirDJV, 2019-03-06:
Based on the [corrmap/Artifact Correction with ICA tutorial](https://mne-tools.github.io/dev/auto_tutorials/plot_artifacts_correction_ica.html#sphx-glr-auto-tutorials-plot-artifacts-correction-ica-py) I'm doing this:
`this_ica = ICA(n_components=epochs.info["nchan"] - 1, method="fastica", random_state=42, max_iter=500).fit(epochs)`

`epochs` is my [`autoreject`](http://autoreject.github.io/index.html) corrected epochs after removal of bad channels (bridged/high percentage of interpolated data or bad data) and was average referenced. Therefore, `n_components=epochs.info["nchan"] - 1` will get different number of channels/components. 

The actual error that I get is: 

```
  File "<ipython-input-37-c3bcced40347>", line 4, in <module>
    show=False, threshold = 0.7)

  File "<string>", line 2, in corrmap

  File "C:\Users\Amir\Anaconda3\envs\mne\lib\site-packages\mne\utils.py", line 952, in verbose
    return function(*args, **kwargs)

  File "C:\Users\Amir\Anaconda3\envs\mne\lib\site-packages\mne\preprocessing\ica.py", line 2495, in corrmap
    nt, mt, s, mx = _find_max_corrs(all_maps, target, threshold)

  File "C:\Users\Amir\Anaconda3\envs\mne\lib\site-packages\mne\preprocessing\ica.py", line 2319, in _find_max_corrs
    all_corrs = [compute_corr(target, subj.T) for subj in all_maps]

  File "C:\Users\Amir\Anaconda3\envs\mne\lib\site-packages\mne\preprocessing\ica.py", line 2319, in <listcomp>
    all_corrs = [compute_corr(target, subj.T) for subj in all_maps]

  File "C:\Users\Amir\Anaconda3\envs\mne\lib\site-packages\mne\utils.py", line 2818, in compute_corr
    return (np.dot(X.T, Y) / float(len(X) - 1)) / (x_sd * y_sd)

ValueError: shapes (31,) and (30,29) not aligned: 31 (dim 0) != 30 (dim 0)
```

jona-sassenhagen, 2019-03-06:
Maybe someone else will, but I will not write code to handle ICA objects with different montages ... you will have to use the same montage for all. 

I will however write a better error message.

agramfort, 2019-03-06:
did you consider interpolating the missing channel?

>

jona-sassenhagen, 2019-03-06:
That would be my preferred way of handling this, too.

AmirDJV, 2019-03-06:
Even though you know that the channel/s is/are almost completely interpolated? 
You would run the ICA -> exclude ICs -> remove the channel/s?

jona-sassenhagen, 2019-03-06:
I would interpolate -> run ICA.

AmirDJV, 2019-03-06:
Even though 90%-100% of your data is interpolated? 

And thank you for the prompt responses :)

jona-sassenhagen, 2019-03-06:
>Even though 90%-100% of your data is interpolated?

Sorry, I don't know what you mean with this?

jona-sassenhagen, 2019-03-06:
In principle, we could do something like:

- implementing `drop_channels` for ICA objects, and simply internally drop any channel missing in one instance for all instances in corrmap. However, you might easily end up in a situation where across many subjects, almost all channels are bad, or even only the critical ones (eg EOGs), so that the matching procedure is pointless
- internally construct the all-vs-template correlation by correlating only the overlapping subset. This would in principle work; in practice, it might run into the same problems as above. I don't think the original procedure implements this.

@sappelhoff

AmirDJV, 2019-03-06:
In autoreject I can use the the reject log to see which epochs in each channel were [good/bad/bad+interpolated](http://autoreject.github.io/generated/autoreject.RejectLog.html?highlight=labels%20contains%20integers%20encode%20channel%20given%20epoch%20good%20value%20bad%20bad%20interpolated). Thus, I can calculate how much of my data was bad/bad+interpolated for each channel. Methodically, I don't think that it's valid to use channels which had high percent of bad/bad+interpolated data. As It's doesn't reflect what "actually" happened.

jona-sassenhagen, 2019-03-06:
Generally, I'm not sure how autoreject and ICA are expected to interact. @jasmainak 

>Methodically, I don't think that it's valid to use channels which had high percent of bad/bad+interpolated data. As It's doesn't reflect what "actually" happened.

Well if you interpolate a lot, you are losing some noise/variance, so your DFs will be underestimated for some purposes. But if you use nonparametric stats, that should not be a problem right?

AmirDJV, 2019-03-06:
> Generally, I'm not sure how autoreject and ICA are expected to interact.

My pipeline/line of thought comes from [here (@jasmainak article)](https://www.frontiersin.org/articles/10.3389/fnins.2018.00530/full) and [here (eeglab)](https://sccn.ucsd.edu/wiki/Makoto%27s_preprocessing_pipeline). They advise to reject and interpolate bad epochs/channels before ICA. However, being somewhat newbie I might be doing it backwards :) 

> implementing drop_channels for ICA objects, and simply internally drop any channel missing in one instance for all instances in corrmap. However, you might easily end up in a situation where across [...]

I agree with you. Running corrmap listwise-ish might result in loss of huge amounts of data. Especially when running corrmap on many subjects.  

In my head - when I used corrmap, I somewhat expected something close to this:
> internally construct the all-vs-template correlation by correlating only the overlapping subset. This would in principle work; in practice, it might run into the same problems as above. I don't think the original procedure implements this

My thought was that it will just match the IC with the other ICs, if that IC is build upon channels that doesn't exist in subject X it will just return a missing match. Why do you think that it will run into the same problem? It's a more pairwise-ish approach. 

> Well if you interpolate a lot, you are losing some noise/variance, so your DFs will be underestimated for some purposes. But if you use nonparametric stats, that should not be a problem right?

I think that in my case it might be problematic. My final analysis will be connectivity between two caps. Using fully interpolated channel doesn't sound good.

jona-sassenhagen, 2019-03-06:
>They advise to reject and interpolate bad epochs/channels before ICA

Well if you do that, you should be fine right? All your montages will be the same.

>I think that in my case it might be problematic. My final analysis will be connectivity between two caps. Using fully interpolated channel doesn't sound good.

Simple solution would be to drop the interpolated channels after you've used the ICA, right?

AmirDJV, 2019-03-06:
> Well if you do that, you should be fine right? All your montages will be the same.

As long as I don't drop bad channels - yes ;) 

> Simple solution would be to drop the interpolated channels after you've used the ICA, right?

You rejected my idea to run the ICA -> exclude ICs -> remove the channel/s ;) 

However, when I think of it, running ICA with fully interpolated channels, wouldn't result with ICs that don't actually reflect my data or might cause excluding ICs that are not actually artifacts? 

Again, thank you for the support!

jona-sassenhagen, 2019-03-06:
>You rejected my idea to run the ICA -> exclude ICs -> remove the channel/s ;)

No, that was a misunderstanding - I don't have an opinion on removing post corrmap/applying the ICA solution.

>However, when I think of it, running ICA with fully interpolated channels, wouldn't result with ICs that don't actually reflect my data or might cause excluding ICs that are not actually artifacts?

I would not apply the ICA solution from interpolated data to uninterpolated data. But if you fit on interpolated and apply to interpolated, no, where would be the problem?

AmirDJV, 2019-03-06:
> I would not apply the ICA solution from interpolated data to uninterpolated data. But if you fit on interpolated and apply to interpolated, no, where would be the problem?

Sure, once I interpolated my data, I'm using only it. I think that it all comes from my problem to trust channels that are almost fully interpolated. Which is why I wanted to use corrmap with ICAs build from different number of components/channels.

jona-sassenhagen, 2019-03-06:
Interpolation simply means taking a weighted mean of the neighbours right? It's not some arcane procedure that creates unpredictable artefacts. So I personally don't see much reason for concern, besides for a reduction in true DFs/increased correlation of channels/rank deficiency.

AmirDJV, 2019-03-06:
I think that @jasmainak would probably have a better answer than me. However, based on this from his [article](https://www.frontiersin.org/articles/10.3389/fnins.2018.00530/full) and from my understanding that each time it only interpolate one channel, I don't necessarily think that it's based on the weighted mean of the neighbors.

> Autoreject is an unsupervised algorithm which minimizes the cross-validation error, measured by the Frobenius norm between the average signal of the training set and the median signal of the validation
> set.

closed 2019-03-06 by larsoner

jasmainak, 2019-03-06:
> However, based on this from his article and from my understanding that each time it only interpolate one channel, I don't necessarily think that it's based on the weighted mean of the neighbors.

@jona-sassenhagen is correct. It is indeed a weighted mean of the other channels. However, when doing this, the bad channels need to be excluded which is why you need to recompute this for every epoch. @AmirDJV any suggestions to improve the doc in autoreject website is welcome :)

AmirDJV, 2019-03-07:
@jona-sassenhagen and @jasmainak happy to be corrected and thank you for your help! :) 

@jasmainak I will be more than happy to give back :)

PR #6036 (2019-03-06): corrmap better error when instances have different channels
Closes  #3891

@AmirDJV