mne-tools/mne-python #3893: Add `pick_channels` to montage classes

choldgraf, 2017-01-10:
mentioned in #3728 

btw I probably shouldn't be the one to implement this PR as I don't have much experience with the philosophy behind montages. Would a `pick_channels` method  just change the `selection` and `ch_names` attributes? And in that case it should probably be a mixin that both `DigMontage` and `Montage` import from.

agramfort, 2017-01-11:
I would go for a BaseMontage class that implements a pick_channels method
and a __getitem__ to be able to do montage[:128]

choldgraf, 2017-01-11:
That sounds good to me. Is there any clear place in the docs where montage objects are explained? E.g., when to use montage vs. digmontage? And maybe explaining the proper way to use the objects after this PR is done? Not sure if that's something that comes up as a confusion point but it couldn't hurt. I might be able to spend time doing this at the docathon in March.

Eric89GXL, 2017-01-11:
I have wondered lately if we should get rid of `Montage` and `DigMontage` altogether and just make a `Digitization` class. Really both `Montage` and `DigMontage` provide positions in space of some subset of: anatomical points, cHPI, electrodes, and extra points. We already have a complete way to store this information, and it's given by `info['dig']`. Each of `Montage` and `DigMontage` have bits of this functionality but are incomplete -- they aren't necessary careful or consistent in treatment of coordinate frames, channel names, etc. In the long run I think we'd benefit from a major refactoring here.

agramfort, 2017-01-11:
you see more the big picture here. No strong feeling but I would not really
know what exactly needs to be done.

choldgraf, 2017-01-11:
yeah I'm in the same boat, if you refactor some of this code under the hood I'm happy to help write some kind of tutorial that explains new behavior.

Eric89GXL, 2017-01-11:
> No strong feeling but I would not really know what exactly needs to be done.

Yeah that's part of the problem. The distinction between `Montage` and `DigMontage` comes up a lot, and the use cases aren't entirely clear so people seem to have trouble keeping them straight.

We currently have two functions, `read_montage` and `read_dig_montage`, that give rise to `Montage` and `DigMontage`, both of which overlap in terms of data they can contain (channel info).

Here is a starting proposal:

* Create a `Digitization` class. This would be a light wrapper around `list` that essentially becomes what is in `info['dig']`. Additional functions
    * Provide function for transforming points to head coord frame.
    * Provide function for computing `dev_head_t`.
* Deprecate `read_montage` and `read_dig_montage` in favor of `read_digitization`, which can read both standard montages and individualized digitizations.
* Deprecate `set_montage` function in favor of `set_digitization`.

@teonbrooks you worked on these classes a lot, WDYT?

choldgraf, 2017-01-11:
What is the conceptual difference between something like the montage classes, and the way that the `nipy` ecosystem (e.g. `nibabel`) handles coordinate frames etc? Would it be worth trying to create resemblances in the API between the two? Or is the usecase too different

Eric89GXL, 2017-01-11:
Maybe. AFAIK they provide transformation matrices to get between frames -- we do this in some places like `info['dev_head_t']`. But I'm not sure if they have equivalent use cases to what we're going for here, which is to provide locations of specific points in multiple coordinate frames (sometimes the same point in multiple frames, or different points in different frames). Plus we already have something that does this tracking of locations, frames, and point types -- `info['dig'`] is a simple list of dicts that is used all over our code to do this, so there are advantages to staying close to this format.

choldgraf, 2017-01-11:
cool - I agree it sounds like a big PITA to change the API and would probably confuse users. Maybe the answer is just some nicely-written tutorials and guides etc. (after a refactor if that's gonna happen)

Eric89GXL, 2017-01-11:
You mean moving to a nipy-like change wouldn't be worth it, but a `Digitization`-like refactoring (what I've proposed) would be?

choldgraf, 2017-01-11:
I think that it if changes an API that is used in lots of other places in MNE, or that lots of other users are already used to, then probably not worth it. If it's more under-the-hood that wouldn't require a shift in how people conceptualize the code, but that streamlines things and reduces redundancy, while making it easier to add better features, then I think it's worth it. It sounds like a full nipy-like refactor would be the former, while your idea of refactoring is more the latter. WDYT?

Eric89GXL, 2017-01-11:
Mine would be a bit of the former, too. But it would unify how we treat things in a nice way. I don't see any real advantages to doing what `nipy` does, I don't think people cross MNE and `nipy` enough to justify the extra work.

agramfort, 2017-01-11:
I am not against a single class but to me digitization refers to measured
locations and there can be more than the number of channels (eg dig points
on the face etc.) At least DigMontage and Montage were conceptually
different since using standard montage restricted to channel locations
differs from a arbitrary number of dig points obtained eg with a polhemus
system

Eric89GXL, 2017-01-11:
> At least DigMontage and Montage were conceptually different since using standard montage restricted to channel locations differs from a arbitrary number of dig points obtained eg with a polhemus system

This concept is nice, yeah. The trade-off we make for keeping it is that people get confused having two classes for storing the same type of information (channel/fiducial/hpi/other point locations in some coordinate frame) that both end up affecting similar things (e.g., layouts while plotting or forward solution calculation). We could keep this distinction at the function level (keep `read_montage` for standard and `read_dig_montage` or `read_digitization` for measured) but return a `Digitization` in both cases -- would that work? If it's still not worth unifying at least the return class I can live with it.

If we can't unify the return type using a new class, a more pressing problem is that every time I go to modify the `DigMontage` class, I have to think about or implement some hacky workaround for the fact that `DigMontage` does not provide as rich a representation as what the `info['dig']` list-of-dict already gives. From looking at the code `DigMontage` seems like it was designed primarily to accommodate other systems and their conventions, but in doing so, ended up with representations that are not as general as what we already have available in `info['dig']`. I'd like to at least `DigMontage` to return something `info['dig']`-like more directly for data from other systems -- then all the things it's supposed to do (convert to head coordinates, compute dev_head_t, save/read, set `info['dig']`), become clear instead of complex.

choldgraf, 2017-01-11:
well I'm happy with whatever you guys go with, since I have very little experience w/ the montage classes. I can say that as a new user it was very confusing to me how to use them. But I think that can be addressed with documentation etc. I'm not going to weigh heavily in on the API discussion though, since I don't think I'm the right user to have in mind here.

teonbrooks, 2017-01-12:
only had a few moments to glance over this. I am for a `Digitization` class, I think. It could be a clean way to handle all of this. 
The initial reason for the schism in functionality deals with the name and usage of montage, which is fixed (cf. https://github.com/mne-tools/mne-python/issues/1703, https://github.com/mne-tools/mne-python/issues/1954#issuecomment-95353841).

I'm MIA for the next 48 hours. I will chime back in here in a few days

agramfort, 2017-01-12:
>
> We could keep this distinction at the function level (keep read_montage
> for standard and read_dig_montage or read_digitization for measured) but
> return a Digitization in both cases -- would that work?
>
> yes I think so. API stays explainable and code maintenance is easier as
there is a single class.

Eric89GXL, 2017-01-12:
> code maintenance is easier as there is a single class.

Yep, and more importantly I think the representation and information stored/provided by the class will be more complete, as it will mimic `info['dig']`.

I'll give this a shot sometime after release, since it's not a huge priority. I'll also retitle and modify the description.

Eric89GXL, 2017-06-07:
Something else to consider:

[MGH Easycap 70-channel translation into 10-20](http://www.nmr.mgh.harvard.edu/meg/manuals/EEG_Cap_Instructions.pdf)

Looking into the 60-channel montage now with @SherazKhan.

larsoner, 2018-10-04:
This one won't make 0.17, moving to 0.18

massich, 2019-08-15:
cross-reference https://github.com/mne-tools/mne-python/issues/3893#issuecomment-271925975 and #6461

larsoner, 2019-09-21:
Closing this since hopefully most of this has been taken care of by the last set of PRs, and we are tracking things in #6461 now

closed 2019-09-21 by larsoner