mne-tools/mne-python #3897: Plotting epochs looping trough subjects

Em-Fl, 2017-01-11:
I am really new to mne and to python, so sorry if it's a very simple question.

I seem to be unable to plot epochs or raw data by looping trough my subjects data without plotting every subject at once, this means that when I run the loop I get all of my subjects data graphed at the same time which consumes a lot of memory.

I've tried with raw input and plt.close, but it doesn't work.
Has anyone encountered a similar problem? some guidance for a beginner?

jaeilepp, 2017-01-12:
I'm not sure I understand your problem, but maybe passing ``block=True`` to the plotter does it.

Eric89GXL, 2017-02-14:
closing since this seems to have been resolved

closed 2017-02-14 by Eric89GXL