mne-tools/mne-python #3900: Anaconda cloud support?

christianbrodbeck, 2017-01-12:
Would it be worth adding Anaconda cloud support? I think the main advantage would be better dependency management. Installation and upgrade instructions for users would reduce to:

```
$ conda config --append channels our_channel_name
$ conda install mne
$ conda update mne
``` 

The way to get it to work would be to hosts mne and pysurfer on a new channel. Adding our own channel would involve maintaining a [meta.yaml file](http://conda.pydata.org/docs/building/meta-yaml.html) and running the build/upload commands at release (which could probably be scripted).

It might be even simpler to add them to [conda-forge](https://conda-forge.github.io/) but I have not tried that yet. It looks like it would build the distributions automatically from the source archive on the PYPI, and some of the dependencies like nibabel are already on conda-forge. If you agree I would give it a try.

CC @Eric89GXL

agramfort, 2017-01-12:
personally for me pip works well enough for a pure python package.

we add some maintenance overhead for little gain.

my 2c

Eric89GXL, 2017-01-12:
I would try conda-forge first rather than having our own channel. I expect it would be easier to integrate into the conda ecosystem that way. But we should check and see what the additional overhead is. I expect the conda-forge route would be one or two extra steps (5-10 minutes?) during each release, which wouldn't be bad.

christianbrodbeck, 2017-01-12:
@Eric89GXL agreed, I will explore that route.

@agramfort I did a small workshop a few weeks ago with people using Macs and Windows and the installation was not as smooth as I would have hoped... Otherwise I would not be spending time on it either...

Eric89GXL, 2017-01-12:
Argh... what problems did you hit? If it's people doing 3.5 vs 2.7 we now at least just recommend 2.7. Can you see if the [updated (yesterday) install docs](http://mne-tools.github.io/dev/install_mne_python.html) might have helped already?

christianbrodbeck, 2017-01-12:
@Eric89GXL We only did 2.7. It was mostly due to the outdated version of Mayavi on anaconda back then, which required manually downgrading multiple packages which had introduced incompatible changes (discussed in #3723). I think the new install docs will work again because Mayavi has been updated. However, now I will have to send people a new list of conda requirements to install, and it would be easier to tell them to just use `$ conda update mne` occasionally...

Eric89GXL, 2017-01-12:
I think if you tell them to do `conda install numpy scipy mayavi` it will upgrade all of them

Eric89GXL, 2017-01-12:
(and the necessary dependencies)

christianbrodbeck, 2017-01-15:
I'll have to look into it more... but unfortunately it does not look like it's that simple, the mayavi update does not seem to enforce the necessary dependencies to update...

christianbrodbeck, 2017-02-09:
@Eric89GXL I assume you're also fine with being a maintainer for mne?

Should we add a dependency on PySurfer? Or Mayavi? Since they're on conda it won't add hassle to the installation, but will make installation more bloated for people that don't use those features. But even if not PySurfer, Mayavi seems to be used by quite a few functions in mne?

Eric89GXL, 2017-02-10:
Yeah feel free to add me. Regarding PySurfer, I would say yes, add as a dependency since we need it for 3D. @agramfort WDYT?

agramfort, 2017-02-10:
Ok for me

Eric89GXL, 2017-02-14:
I think @christianbrodbeck is on this at the conda repos so I'll close this

closed 2017-02-14 by Eric89GXL