mne-tools/mne-python #3901: Error when set channel types in raw file

sarwatfatimam, 2017-01-13:
Hi, 

I am trying to set channel types in the raw file using: 
raw.set_channel_types({'EEG061':'misc', 'EEG062':'eog', 'EEG063' :'ecg', 'EEG064':'misc'} ) 

However, I get the following error:
RuntimeError: Cannot change channel type for channel EEG063 in projector "Average EEG reference"

The average eeg reference is off when I print out raw.info. I have added the info below. Can someone please help me with this? 

<Info | 29 non-empty fields
    acq_pars : 'str | 21833 items
    bads : 'list | 0 items
    buffer_size_sec : 'numpy.float64 | 1.0
    ch_names : 'list | MEG0113, MEG0112, MEG0111, MEG0122, MEG0123, MEG0121, ...
    chs : 'list | 404 items (EEG: 74, STIM: 3, GRAD: 204, CHPI: 9, MISC: 12, MAG: 102)
    comps : 'list | 0 items
    custom_ref_applied : 'bool | False
    description : 'str | 36 items
    dev_head_t : 'mne.transforms.Transform | 3 items
    dig : 'list | 137 items
    events : 'list | 1 items
    experimenter : 'str | 3 items
    file_id : 'dict | 4 items
    filename : 'str | F:\Analysi.../run_01_sss.fif
    highpass : 'float | 0.0 Hz
    hpi_meas : 'list | 1 items
    hpi_results : 'list | 1 items
    hpi_subsystem : 'dict | 2 items
    line_freq : 'float | 50.0
    lowpass : 'float | 356.3999938964844 Hz
    meas_date : 'numpy.ndarray | 2009-04-09 16:04:14
    meas_id : 'dict | 4 items
    nchan : 'int | 404
    proc_history : 'list | 1 items
    proj_id : 'numpy.ndarray | 1 items
    proj_name : 'str | 11 items
    projs : 'list | Average EEG reference: off
    sfreq : 'float | 1100.0 Hz
    subject_info : 'dict | 6 items
    acq_stim : 'NoneType
    ctf_head_t : 'NoneType
    dev_ctf_t : 'NoneType
    kit_system_id : 'NoneType
    xplotter_layout : 'NoneType
>

jaeilepp, 2017-01-13:
The average reference projector is already computed even though it is off, so it needs to be recomputed to match the data. You can use ``raw.del_proj`` to remove it (another option is to read in the data with ``add_eeg_ref=False``). Be sure to compute it after changing the channel types with ``raw.set_eeg_reference()``.

jaeilepp, 2017-01-13:
Here's an example http://mne-tools.github.io/mne-biomag-group-demo/auto_scripts/05-make_epochs.html#sphx-glr-auto-scripts-05-make-epochs-py

sarwatfatimam, 2017-01-13:
It is working now. Thankyou.

closed 2017-01-13 by sarwatfatimam