mne-tools/mne-python #3902: How to avoid memory allocation error of ICA?

lix90, 2017-01-13:
Hi all,

These days I wanted to move from matlab (EEGLAB) to python (MNE), so I started to learn MNE several days ago. **But I run into an `error" can't allocate region` over and over again when I run ICA (fastica) on both raw data and epoched data.** I tried band-pass filtered at 1-40Hz and resampled at 100-250Hz, and I also set the parameter `decim` to 3 or 5.

This is the raw dataset when preloaded:
`n_channels x n_times : 64 x 792890 (1585.8 sec), ~387.3 MB, data loaded`

My computer is a 2015 Macbook pro with 16GB RAM. I never came across this kind of memory error when I used EEGLAB's extended-infomax ICA with the same datasets both in windows or linux PC before. But I have not tried this on my Macbook. It's hard to imagine and confused me that it can consume so much memory (size=18446744073183428608).

I found that there is few information about the issue both in the MNE-Python website and in the email-archive. Any suggestion about avoiding such memory allocation error will be greatly appreciated.

```
Python(51320,0x7fff75f54000) malloc: *** mach_vm_map(size=18446744073183428608) failed (error code=3)
*** error: can't allocate region
*** set a breakpoint in malloc_error_break to debug
```

agramfort, 2017-01-13:
what version of scikit-learn and mne do you use?

lix90, 2017-01-13:
mne: 0.13.1
sklearn: 0.18.1

agramfort, 2017-01-13:
ok so it's recent.

did you try the decim parameter in ICA?

lix90, 2017-01-13:
Yeah, I tried `decim`, 3 or 5. It would work finely when the sample frequency was quite low and `decim` was not None. But I suspect the results of the decomposition. I consulted the documentation of MNE and I found that `decimation` is similar with resampling in my amteur eyes.

agramfort, 2017-01-13:
I suspect your result will not be too much affected by decim

lix90, 2017-01-13:
Is it ok to have sample frequency below 100Hz and simultaneously set `decim` to 3 or 5 when run ICA to detect EOG/ECG artifacts? I am also confused with the `decim` parameter. Is there big difference between 3 or 5 of decimation? Because I saw 3 or 5 were frequently used in examples or researchers' scripts.

I will try more parameters and datasets.

agramfort, 2017-01-13:
you need to try on your data.

but I am really surprised you need so much RAM. It should not make more
than 3 copies of the data.

jmontoyam, 2017-01-13:
Hi @lix90, 
This error is very strange, I have never encountered such an error before.
In order to help you, I would like to try to reproduce this error, could you please share the file
that you are trying to load, as well as a tiny script to reproduce the error
message?

lix90, 2017-01-13:
Sure, tomorrow morning I will upload it (Because it's midnight in China :-) ). 
Thanks for the helps @agramfort @jmontoyam 

There are two types of dataset: one is the raw eeglab dataset merged from 5 seperate datasets (Old); the other is the raw brainvision dataset (New). Both are from an experiment of emotion regulation.

lix90, 2017-01-14:
@jmontoyam The data and codes are uploaded [there](https://drive.google.com/drive/folders/0B9Ht1ZLPrqYZaktOUE0taThOYlU?usp=sharing)

I tried this on jupyter notebook and ipython in emacs. Both resulted in such memory allocation error. You can change sample frequency, reject threshold, decimation, and filter parameters. It is really strange that the size allocated seems similar (for example, size=18446744066833018880 yesterday; size=18446744073183428608 today). strange 184467440************ :(

agramfort, 2017-01-14:
your code worked smoothly on my laptop with 8GB of RAM and maybe other apps
open.

I used "1canyu_merge.set"

jmontoyam, 2017-01-14:
Hi @lix90,

I have tried to reproduce your error message, but in my machine (Linux x64 with 16GB of RAM) your code works perfectly. I used "s3-er.vhdr" and "1canyu_merge.set", and in both cases everything worked smoothly.
I'm using Python, as well as all the scientific Python packages available in Anaconda, what about you?

lix90, 2017-01-14:
Oh, that's weired. I just use Python2.7 which is installed by Homebrew. All modules were installed by pip. The error is real but not that stable, because “the threshold” of parameters causing this error would change next time. And I use python in terminal (iterm), python shell in Emacs, and jupyter notebook as well. It is really weired. Maybe there is something wrong with my configuration. I don't know. But the error with such strange size is unusual.

Did you @agramfort @jmontoyam change sfreq, or decim, or other parameters? Raw data with sfreq=250Hz and decim=3 consistently end up with such error.

lix90, 2017-01-14:
Below is the error message:

```
Process:               Python [66268]
Path:                  /usr/local/Cellar/python/2.7.12_2/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python
Identifier:            Python
Version:               2.7.12 (2.7.12)
Code Type:             X86-64 (Native)
Parent Process:        bash [65799]
Responsible:           iTerm2 [63216]
User ID:               502

Date/Time:             2017-01-15 00:14:43.232 +0800
OS Version:            Mac OS X 10.11.5 (15F34)
Report Version:        11
Anonymous UUID:        269DBE64-809E-BB92-5C38-40A1FB6EBC42

Sleep/Wake UUID:       8C7B7668-095C-4636-9826-045A95DA0D96

Time Awake Since Boot: 400000 seconds
Time Since Wake:       4300 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000

VM Regions Near 0:
--> 
    __TEXT                 00000001062d3000-00000001062d5000 [    8K] r-x/rwx SM=COW  /usr/local/Cellar/python/2.7.12_2/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libBLAS.dylib                 	0x00007fff904ee625 dgePack_A_NoTran_AVX2 + 121
1   libLAPACK.dylib               	0x00007fff8b960ea4 dgetrf_FactorColumn + 1106
2   libLAPACK.dylib               	0x00007fff8b96154b DGETRF + 1195
3   _flinalg.so                   	0x00000001078092de dlu_c_ + 142
4   _flinalg.so                   	0x0000000107804274 f2py_rout__flinalg_dlu_c + 868
5   org.python.python             	0x00000001062e59b4 PyObject_Call + 99
6   org.python.python             	0x0000000106364e30 PyEval_EvalFrameEx + 26668
7   org.python.python             	0x000000010635e40e PyEval_EvalCodeEx + 1617
8   org.python.python             	0x0000000106368c9e fast_function + 117
9   org.python.python             	0x0000000106364f09 PyEval_EvalFrameEx + 26885
10  org.python.python             	0x000000010635e40e PyEval_EvalCodeEx + 1617
11  org.python.python             	0x0000000106368c9e fast_function + 117
12  org.python.python             	0x0000000106364f09 PyEval_EvalFrameEx + 26885
13  org.python.python             	0x000000010635e40e PyEval_EvalCodeEx + 1617
14  org.python.python             	0x0000000106368c9e fast_function + 117
15  org.python.python             	0x0000000106364f09 PyEval_EvalFrameEx + 26885
16  org.python.python             	0x0000000106368d31 fast_function + 264
17  org.python.python             	0x0000000106364f09 PyEval_EvalFrameEx + 26885
18  org.python.python             	0x0000000106368d31 fast_function + 264
19  org.python.python             	0x0000000106364f09 PyEval_EvalFrameEx + 26885
20  org.python.python             	0x000000010635e40e PyEval_EvalCodeEx + 1617
21  org.python.python             	0x0000000106368c9e fast_function + 117
22  org.python.python             	0x0000000106364f09 PyEval_EvalFrameEx + 26885
23  org.python.python             	0x0000000106368d31 fast_function + 264
24  org.python.python             	0x0000000106364f09 PyEval_EvalFrameEx + 26885
25  org.python.python             	0x0000000106368d31 fast_function + 264
26  org.python.python             	0x0000000106364f09 PyEval_EvalFrameEx + 26885
27  org.python.python             	0x000000010635e40e PyEval_EvalCodeEx + 1617
28  org.python.python             	0x0000000106303b87 function_call + 350
29  org.python.python             	0x00000001062e59b4 PyObject_Call + 99
30  org.python.python             	0x0000000106365697 PyEval_EvalFrameEx + 28819
31  org.python.python             	0x000000010635e40e PyEval_EvalCodeEx + 1617
32  org.python.python             	0x0000000106368c9e fast_function + 117
33  org.python.python             	0x0000000106364f09 PyEval_EvalFrameEx + 26885
34  org.python.python             	0x000000010635e40e PyEval_EvalCodeEx + 1617
35  org.python.python             	0x0000000106368c9e fast_function + 117
[... 309 lines cut ...]
STACK GUARD                       56.0M       11 
Stack                             12.6M       11 
VM_ALLOCATE                       22.3M       53 
__DATA                            6664K      220 
__LINKEDIT                        98.5M      137 
__TEXT                            52.7M      192 
__UNICODE                          552K        2 
shared memory                       12K        4 
===========                     =======  ======= 
TOTAL                              1.2G      665 

Model: MacBookPro11,4, BootROM MBP114.0172.B09, 4 processors, Intel Core i7, 2.2 GHz, 16 GB, SMC 2.29f23
Graphics: Intel Iris Pro, Intel Iris Pro, Built-In
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533642465238412D50422020
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533642465238412D50422020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x152), Broadcom BCM43xx 1.0 (7.21.95.175.1a6)
Bluetooth: Version 4.4.5f3 17904, 3 services, 18 devices, 1 incoming serial ports
Network Service: USB 10/100 LAN, Ethernet, en4
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SM0256G, 251 GB
USB Device: USB 3.0 Bus
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Bluetooth USB Host Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 27.1
```

mmagnuski, 2017-01-14:
It works well for me too. I did not change anything in the notebook apart from file path and py3 compatibility for `filter`.
Can you reproduce the error in a conda virtual environment that matches the versions of packages that you are using?

mmagnuski, 2017-01-14:
(and just BTW - when reading the `.set` file to mne you seem to be dropping a few 'boundary' eeglab events - that won't influence your ICA, but it's better to read those in too, to know where the discontinuities are in your signal)

lix90, 2017-01-15:
I switched my python environment to Python3.5 in Anaconda and uninstall pure Python2.7. Then I created a Python2.7 environment with complete anaconda packages and pip installed MNE.

First run returned the message below:

```
/Users/lix/anaconda3/envs/python2/lib/python2.7/site-packages/sklearn/decomposition/fastica_.py:116: UserWarning: FastICA did not converge. Consider increasing tolerance or the maximum number of iterations.
```

Sencond run in which I set `max_iter=512` still returned the same message.
Third run without `max_iter` parameter still returned the message.

I did not get the memory error with Anaconda, I don't know why. But the fastica warning emerged. I have no background of programming and signal processing or something else related to it, so these all puzzled me. : (

lix90, 2017-01-15:
> when reading the .set file to mne you seem to be dropping a few 'boundary' eeglab events - that won't influence your ICA, but it's better to read those in too, to know where the discontinuities are in your signal

Hi @mmagnuski Thank you for the reminder. I did not pay much attention to it. It seems like the `boundary` events are defaultly dropped. What should I do in order to prevent dropping boundary events? specifying the boundary events like {'boundary': 99} or something like that? I notice the `event_id_func` but I am not clear how to use it right now. BTW, if boundary events are informative, why not keep it in default?

mmagnuski, 2017-01-15:
Yes, you can just add the boundary events to `event_id`, for example:
```python
raw = mne.io.read_raw_eeglab('some_data.set', event_id={'stim': 1, 'resp': 2, 'boundary': 99})
```
(assuming you also have events `stim` and `resp` and are reading raw eeglab data). They could be read by default as 99 for example, but probably that wouldn't be clear to the user - what these `99` events mean.
Another option would be to read boundary events as very short annotations ('bad - discontinuity' for example). This way they could be used when constructing epochs from raw using `reject_by_annotations`.

lix90, 2017-01-15:
Thank you @mmagnuski for such a detailed guide.

According to your description, I looked up the documentation and tried to write a tiny script to implement this procedure. I paste the snippet here just for validation and considering that it might be helpful for MNE & Python beginners just like me.

```
import numpy as np
import mne

fname_eeglab = '/path/to/some_data.set'
events_id = {'stim': 1, 'resp': 2, 'boundary': 99}
bad_duration = 0.5

raw = mne.io.read_raw_eeglab(fname_eeglab, event_id=events_id)
events = mne.find_events(raw)
boundary_events = events[events[:, 2] == events_id['boundary'], :]
n_boundary_events = len(boundary_events)
boundary_onsets = boundary_events[:, 0] / raw.info['sfreq'] - bad_duration / 2
boundary_durations = np.repeat(bad_duration, n_boundary_events)
raw.annotations = mne.Annotations(onset=boundary_onsets,
                                  duration=boundary_durations,
                                  description=['bad - discontinuity'],
                                  orig_time=raw.info['meas_date'])
raw.plot(events=boundary_events)

# ... ... other codes here

epochs = mne.Epochs(raw, events, event_id, tmin, tmax, proj=True,
                    picks=picks_meg, baseline=baseline, reject=reject,
                    reject_by_annotation=True, add_eeg_ref=False)
```

closed 2017-01-16 by lix90