mne-tools/mne-python #3906: label.center_of_mass 'Cannot compute COM with negative values' for MNE labels

jhouck, 2017-01-13:
I ran across this in my data and replicated it using the sample data.  Something simple like:

`subjects_dir = os.getenv('MNE_SAMPLE') + '/subjects'`
`subject = 'sample'`
`label = mne.read_label(filename=subjects_dir + '/' + subject + '/label/lh.BA1.label', subject='sample', color=None)`
`idx = label.center_of_mass(subject=subject, restrict_vertices=True, subjects_dir=subjects_dir, surf='sphere')`

using one of Freesurfer's BA labels seems to work fine (returns 59963).  But the same thing after reading from the annotation like:

`subjects_dir = os.getenv('MNE_SAMPLE') + '/subjects'`
`subject = 'sample'`
`labels = mne.read_labels_from_annot(subject, subjects_dir=subjects_dir, parc='aparc.a2009s', hemi='both')`
`label_names = [label.name for label in labels]`
`label_idx = label_names.index('S_temporal_transverse-rh')`
`label=labels[label_idx]`
`idx = label.center_of_mass(subject=subject, restrict_vertices=True, subjects_dir=subjects_dir, surf='sphere')`

fails with "ValueError: All values must be non-negative and at least one must be non-zero, cannot compute COM".  This seems to be because the private function _center_of_mass in SourceEstimate doesn't want label.values to all == 0, which is the case for all of MNE's labels.  The simple expedient of doing `label.values = np.ones(label.values.shape)` before `label.center_of_mass` seems to work -- for the BA labels the COM vertex lands more or less where you'd expect it to be, but does hug the edge of some of the long, narrow labels.  

There's probably a better way to do this, but I don' t know enough about the vertex numbering to make sense of it.

jhouck, 2017-01-13:
To clarify: anatomical labels.  For functional labels this would be a _really_ terrible idea.

Eric89GXL, 2017-01-13:
The simple expedient of doing label.values = np.ones(label.values.shape)
before label.center_of_mass seems to work -- for the BA labels the COM
vertex lands more or less where you'd expect it to be, but does hug the
edge of some of the long, narrow labels

This is the correct way to get an unweighted or equally weighted average.
Can you make a PR to clarify the error message and or documentation?

jhouck, 2017-01-17:
I'll give it a shot.

Eric89GXL, 2017-02-14:
This has been fixed thanks to @jhouck

closed 2017-02-14 by Eric89GXL