mne-tools/mne-python #3907: ENH: plot_white rotation

Eric89GXL, 2017-01-15:
From Matti on the listserv:

> y’ = lambda^(-1)U^t y
> y’’ = U lambda^(-1)U^t y
> the y’’ data have their original spatial meaning and the values are SNRs or “z scores”.  Since the whitened data are sphered, they can be rotated in any way. Maybe there is even another more interesting rotation than U.

I suspect we do the former but the latter would be more useful I think. @dengemann I think you worked on this the most, do you agree?

agramfort, 2017-01-16:
evoked.plot_white does y'' (ie rotates back) otherwise the whitened evoked
would not look like a butterfly as in
https://mne-tools.github.io/dev/auto_examples/visualization/plot_evoked_whitening.html

Eric89GXL, 2017-01-16:
Ahh okay, good

closed 2017-01-16 by Eric89GXL