mne-tools/mne-python #3908: DISCUSS: GSoC 2017

Eric89GXL, 2017-01-16:
> Organizations can start submitting applications for Google Summer of
> Code 2017 on January 19 (and the deadline is February 9)

We don't need to have everything hashed out by Feb 9, but it would be a good idea if possible to have some ideas on our Wiki so python.org (we apply as a suborg of python.org) can use it in their application to Google.

Anybody have ideas for specific projects, or know of students who would be interested in working on GSoC?

ravaan, 2017-01-19:
Hey! I am a student and am really interested in working on GSoC, I am a beginner and would be very great if you can point where I can start from.

agramfort, 2017-01-20:
@Eric89GXL start from this wiki page: https://github.com/mne-tools/mne-python/wiki/GSOC-Ideas

Regarding ideas, one would be to facilitate access to open EEG/MEG databases so work on datasets module. There is physionet, but much more. Having a consistent API to access multiple data source would be great.

Eric89GXL, 2017-02-14:
Closing as we have the Wiki page up

closed 2017-02-14 by Eric89GXL