mne-tools/mne-python #3917: Brainstorm data not loading?

choldgraf, 2017-01-19:
I'm trying to download the brainstorm data, but it seems to hang when running `get_path`. It prompts for me to accept the agreement, and then...nothing happens. It just sits there (I've left it running for upwards of 20 minutes). Is this expected behavior, and the file is just really large? If not, has anybody else experienced this?

Also it'd be nice to add an extra flag to the `data_path` to accept the license, so that you don't have to do it with `sys.argv`. I can do that in a PR once I figure out the download issue.

agramfort, 2017-01-19:
even with

mne.set_log_level('INFO')

?

Eric89GXL, 2017-01-19:
or more easily `verbose=True` in the call?

I just moved my `bst_phantom_elekta` data and ran `python -c "import mne; mne.datasets.brainstorm.bst_phantom_elekta.data_path(verbose=True)"` locally and it worked fine. This suggests it was a temporary connectivity issue, or local connectivity problem. But if the problem persists from multiple download locations (home, work, etc.) feel free to reopen

closed 2017-01-19 by Eric89GXL

choldgraf, 2017-01-20:
yep it must have been a temporary problem because it worked for me just now too