mne-tools/mne-python #3919: compute_source_psd

annapasca, 2017-01-19:
I got the following error 
operands could not be broadcast together with shapes (22494,327) (7498,327) (22494,327) 

when I run this code 
```
from mne import io
from mne.datasets import sample
from mne.minimum_norm import read_inverse_operator, compute_source_psd

data_path = sample.data_path()
subjects_dir = data_path + '/subjects'

raw_fname = data_path + '/MEG/sample/sample_audvis_raw.fif'
fname_inv = data_path + '/MEG/sample/sample_audvis-meg-oct-6-meg-inv.fif'

raw = io.read_raw_fif(raw_fname, verbose=False)
raw.info['bads'] = ['MEG 2443', 'EEG 053']

inverse_operator = read_inverse_operator(fname_inv)

stc = compute_source_psd(raw, inverse_operator, lambda2=1. / 9., method="dSPM",
                         tmin=0, tmax=120, fmin=4, fmax=100,
                         pick_ori=None, n_fft=2048)

```

agramfort, 2017-01-20:
I can replicate the pb.

this line above

psd = np.zeros((K.shape[0], np.sum(freqs_mask)))

is wrong. It should take into account the number of ori if "is_free_ori and pick_ori is None"

can you open a PR and use this snippet as a test?

annapasca, 2017-01-20:
Hi Alex,

I'm testing and it seems it is solved with the check
```

    if is_free_ori and pick_ori is None:    
        psd = np.zeros((K.shape[0]/3, np.sum(freqs_mask)))
    else:
        psd = np.zeros((K.shape[0], np.sum(freqs_mask)))

```

Ok, I'll open a PR!

jaeilepp, 2017-01-26:
This was fixed with https://github.com/mne-tools/mne-python/pull/3929

closed 2017-01-26 by jaeilepp