mne-tools/mne-python #3921: mne.gui.coregistration()

annapasca, 2017-01-20:
If I read into the dig field of the CTF sample data
```

import mne
import os.path as op
from mne.datasets import spm_face

data_path = spm_face.data_path()
raw_fname = op.join(data_path,
                    'MEG/spm/SPM_CTF_MEG_example_faces1_3D_raw.fif')
raw = mne.io.read_raw_fif(raw_fname)
raw.info['dig']
```

it seems to me that the right set to use the gui is
```

raw.info['dig'][0]['kind'] = 1
raw.info['dig'][1]['kind'] = 1
raw.info['dig'][2]['kind'] = 1
raw.info['dig'][0]['ident'] = 2
raw.info['dig'][1]['ident'] = 1
raw.info['dig'][2]['ident'] = 3
```

is it? 
only with this set the gui shows the fiducials and the right positions

jaeilepp, 2017-01-20:
I believe you're right. The CTF definitions use ids 2 for rpa and 3 for nasion. Fix coming up.

annapasca, 2017-01-20:
great!

closed 2017-03-01 by jaeilepp