mne-tools/mne-python #3926: evoked.plot_white() for raw data

olafhauk, 2017-01-23:
An equivalent of evoked.plot_white() for raw data would be very useful for people who want to compare whitening methods for raw data (e.g. resting state) or empty room data.

Eric89GXL, 2017-01-23:
Could we add an option to raw.plot? That might be easiest

agramfort, 2017-01-23:
how would it look like API wise?

Eric89GXL, 2017-01-23:
You could have `whitening_cov=None` or `cov=None` and if they pass one in, then you whiten the selected channels (and use an appropriate scale for them). But maybe it's not quite as nice because then the scale would be difficult to see.

Come to think of it, we could create a new type of raw plot that is a butterfly plot like `evoked.plot()` but with scroll interactivity. Maybe that would work better.

agramfort, 2017-01-23:
to be consistent it could be a

raw.plot_white function.

and I am thinking that we could improve our raw data viewer to toggle
between butterfly plots and current approach

olafhauk, 2017-01-23:
Such a "butterfly" option (with -1/+1 range highlighted) would be great. It would help identify data segments for which the whitening doesn't work well. 
Alternatively, it could be like the current raw.plot() function, but with an added GFP channel. If one can identify individual channels, this might help with artefact or bad channel detection.

Eric89GXL, 2017-02-20:
> and I am thinking that we could improve our raw data viewer to toggle between butterfly plots and current approach

Actually I don't think this would be too terribly difficult. That's 1) change ylabels, 2) change all offsets to be `0`, and 3) change scale, I think.

So the API changes would be:

1. Add `raw.plot(..., mode='butterfly')`
2. Add `raw.plot(..., cov=None)`
3. Add `raw.plot_white(cov)` which just wraps to `raw.plot(..., cov=None, mode='butterfly')`

I think I'd rather omit step (3) and just reference `raw.plot` in `evoked.plot_white`. Maybe there is a similar way to change `evoked.plot` to work like `evoked.plot_white`, and deprecate `evoked.plot_white`?

dengemann, 2017-02-20:
I'd be happy not to deprecate evoked.plot_white and not overcharge
evoked.plot which is already quite a beast.

agramfort, 2017-02-21:
no objection

PR #4799 (2018-02-28): MRG: Add whitening to plots
Need to:

- [x] fill out docstrings
- [x] add tests
- [x] update whats_new.rst

Will post CircleCI result once it comes up.

Closes #3926.
closed 2018-02-28 by agramfort