mne-tools/mne-python #3928: How can I run spatio_temporal_cluster_1samp_test() within a certain region?

emmachen219, 2017-01-25:
I have a question about creating a mask for spatio_temporal_cluster_1samp_test(). I'm trying to restrict the cluster analysis on the occipital and temporal region for both left and right hemispheres. 
```
# Define spatial_exclude
label_lh = mne.read_labels_from_annot('fsaverage', parc='PALS_B12_Lobes', subjects_dir=subjects_dir,hemi='lh')
label_rh = mne.read_labels_from_annot('fsaverage', parc='PALS_B12_Lobes', subjects_dir=subjects_dir,hemi='rh')
keep_stc = stc_Obj.in_label(label_lh[3] + label_lh[5] + label_rh[6] + label_rh[8])
exclude_lh = ~np.in1d(All_vertices_lh, keep_stc.vertices[0])
exclude_rh = ~np.in1d(All_vertices_rh, keep_stc.vertices[1])
spatial_exclude = np.concatenate([np.where(exclude_lh)[0], np.where(exclude_rh)[0] + len(All_vertices_lh)])

#Run cluster analysis
connectivity = spatial_tris_connectivity(grade_to_tris(4))
X = np.transpose(X, [2, 1, 0])  # X.shape was [5124, 81, 9], vertices * times * subjects
p_threshold = 0.05
t_threshold = -stats.distributions.t.ppf(p_threshold / 2., n_subjects - 1)
T_obs, clusters, cluster_p_values, H0 = clu = \
    spatio_temporal_cluster_1samp_test(X, connectivity=connectivity, n_jobs=2,
                                       spatial_exclude = spatial_exclude,
                                       threshold=t_threshold)

```
However, the results show that there are still clusters outside of the occipital and temporal regions. In the figure attached as an example, you can see from the lateral view, there are clusters at the frontal lobe and at the conjunction between the parietal lobe and temporal lobe; from the ventral view, there are clusters in the subcortical region.

[image]

Eric89GXL, 2017-01-25:
Can you turn this into a SSCCE (http://sscce.org) that uses the sample
dataset? We at least need X (can be a fake example) and the associated
plotting call to be able to replicate this

emmachen219, 2017-01-26:
Thanks very much for your time, Eric! I've updated the code to be at least self-contained. I'm afraid I cannot find a way to make it short at the moment though.

Eric89GXL, 2017-02-03:
I see the problem. Your `X` and `connectivity` are defined on `fsaverage`'s ICO-5 source space, which FYI conveniently has its vertices-in-use given by `fsaverage_vertices = [np.arange(10242), np.arange(10242)]`.

Your labels are from your specific subject (here `sample`) source space.

You need to use labels from `fsaverage` when computing the `exclude`. It works for me when changing these lines:
```
### Define spatial_exclude on fsaverage
label_V1_lh = mne.read_label(data_path + '/subjects/fsaverage/label/lh.V1.label')
label_V1_rh = mne.read_label(data_path + '/subjects/fsaverage/label/rh.V1.label')
label_V2_lh = mne.read_label(data_path + '/subjects/fsaverage/label/lh.V2.label')
label_V2_rh = mne.read_label(data_path + '/subjects/fsaverage/label/rh.V2.label')

fsaverage_vertices = [np.arange(10242), np.arange(10242)]
exclude_lh = ~(np.in1d(fsaverage_vertices[0], label_V1_lh.vertices) |
               np.in1d(fsaverage_vertices[0], label_V2_lh.vertices))
exclude_rh = ~(np.in1d(fsaverage_vertices[1], label_V1_rh.vertices) |
               np.in1d(fsaverage_vertices[1], label_V2_rh.vertices))
spatial_exclude = np.concatenate([np.where(exclude_lh)[0], np.where(exclude_rh)[0] + len(fsaverage_vertices[0])])
```
It also helps to change `event_id=1` to `event_id=4` so it's a L/R visual contrast.

closed 2017-02-03 by Eric89GXL