mne-tools/mne-python #3932: BUG: freesurfer environment?

kingjr, 2017-01-25:
I get a `command not found: mri_watershed' error` when running `mne.bem.make_watershed_bem` on a mac OS X Yosemite 10.10.4 + conda python 2.7 with mne 0.13.1 install through pip.

Yet, the same command can be called directly from the shell, and can be called through subprocess:i.e. this works
```python
import subprocess
subprocess.check_output(['mri_watershed', '-useSRAS', '-surf', ...)
```

I tried to manually add `FREESURFER_HOME` to `os.environ` and do the same via `mne.utils.set_config` without success. 

Any idea?

maybe related to #2814

Eric89GXL, 2017-01-26:
Can you try on latest `master`?

kingjr, 2017-01-26:
Problem solved with latest master

closed 2017-01-26 by kingjr