mne-tools/mne-python #3933: check `n_components > max_pca_components` in read_ica fails

jona-sassenhagen, 2017-01-26:
A friend's ICA object can't be read because `n_components` is stored as `None` and `max_pca_components` is an `int`. Can anybody think of where this one is coming from?

I guess the check could be circumvented, but I'm not sure that's the optimal fix.

@dengemann 

```
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-3-9879665cf5a0> in <module>()
----> 1 mne.preprocessing.read_ica("/Users/jona/Downloads/ss0018_ET_EEG-ica.fif")

/Users/jona/tools/mne-python/mne/preprocessing/ica.py in read_ica(fname)

/Users/jona/tools/mne-python/mne/utils.py in verbose(function, *args, **kwargs)
    707         with use_log_level(verbose_level):
    708             return function(*args, **kwargs)
--> 709     return function(*args, **kwargs)
    710 
    711 

/Users/jona/tools/mne-python/mne/preprocessing/ica.py in read_ica(fname)
   1904     ica_init = dict((k, v) for k, v in ica_init.items()
   1905                     if k in _get_args(ICA.__init__))
-> 1906     ica = ICA(**ica_init)
   1907     ica.current_fit = current_fit
   1908     ica.ch_names = ch_names.split(':')

/Users/jona/tools/mne-python/mne/preprocessing/ica.py in __init__(self, n_components, max_pca_components, n_pca_components, noise_cov, random_state, method, fit_params, max_iter, verbose)

/Users/jona/tools/mne-python/mne/utils.py in verbose(function, *args, **kwargs)
    707         with use_log_level(verbose_level):
    708             return function(*args, **kwargs)
--> 709     return function(*args, **kwargs)
    710 
    711 

/Users/jona/tools/mne-python/mne/preprocessing/ica.py in __init__(self, n_components, max_pca_components, n_pca_components, noise_cov, random_state, method, fit_params, max_iter, verbose)
    235 
    236         if max_pca_components is not None and \
--> 237                 n_components > max_pca_components:
    238             raise ValueError('n_components must be smaller than '
    239                              'max_pca_components')

TypeError: unorderable types: NoneType() > int()
```

dengemann, 2017-01-26:
Can we reproduce this in a test? I thought our IO coverage is not so bad.

Eric89GXL, 2017-01-26:
Or is it possible the ICA object is very old? Maybe we had an old bug

jona-sassenhagen, 2017-01-26:
It's supposedly from 0.13 ... I'll try to come up with a reproducible example. I'm a bit flabbergasted, but it sounds like something we've previously sorted out!

dengemann, 2017-01-26:
Yes stuff like this can happen with old objects. We don't really explicitly
support old objects ...

timmidee, 2017-02-01:
As this is my ICA object I figured I would just to briefly mention that the ICA objects are made in the same script as they are loaded in, i.e. not made in some older version of MNE than the one that I'm trying to read them with...

dengemann, 2017-02-01:
Ah. Can you share some code that shows this issue on the MNE sample data?

timmidee, 2017-02-01:
Sorry, I was about to update that comment with the information that I ran the ICA on a concatenation of two datasets (from the same participant, same setup, same session, etc... just two files). Not sure that is relevant, but I'll leave that judgement to you!

I'll try with the example data. Can you give me a quick hint where/how to easily find that data?

dengemann, 2017-02-01:
Hi Tim,

have a look at any example from our website. If you run any of those the
data will be downloaded if they aren't already on your local drive.

timmidee, 2017-02-01:
Thanks, I tried and I get no error when using the example data...

On the other hand, when I use my own data and print back ica.n_components and ica.max_pca_components right before saving the object, I get 

None
62

So the i/o seems to be doing its work just fine, in a sense that it is reading and writing properties of the ICA object. It appears to be the ICA object itself that has properties that are not allowed as a combination of arguments to the init of a new ICA object (as far as I can tell by the error messages).

hoechenberger, 2017-02-06:
I can reliably reproduce this issue using the example data with the following code:
```python
# -*- coding: utf-8 -*-

from __future__ import division , print_function

import os
import mne
from mne.datasets import sample

data_path = sample.data_path()
output_path = os.path.normpath('/tmp')

raw_fname = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw.fif'
event_fname = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw-eve.fif'
tmin, tmax = -0.2, 0.5
event_id = dict(aud_l=1, vis_l=3)

# Setup for reading the raw data
raw = mne.io.read_raw_fif(raw_fname, preload=True)
events = mne.read_events(event_fname)

# Read epochs
epochs = mne.Epochs(raw, events, event_id, tmin, tmax, proj=True,
                    decim=4, baseline=None, preload=True, )

eeg_epochs = epochs.copy().pick_types(meg=False, eeg=True)

ica = mne.preprocessing.ICA(n_components=None, random_state=1)
ica.fit(eeg_epochs)
ica.save(fname=os.path.join(output_path, 'test_ica-ica.fif'))

# This is going to fail.
ica = mne.preprocessing.read_ica(fname=os.path.join(output_path, 'test_ica-ica.fif'))
```

PR #3967 (2017-02-24): FIX: Check if n_components is None
The comparison fails if either max_pca_components or
n_components (or both) is None.

Closes #3933.
commit baffd5af (2017-02-24): FIX: Check if n_components is None (#3967)

* FIX: Check if n_components is None

The comparison fails if either max_pca_components or
n_components (or both) is None.

Closes #3933.

* TST: Add test

* TST: Move ICA tests to proper location

* Update whats_new
closed 2017-02-24 by Eric89GXL