mne-tools/mne-python #3939: What's the closest thing that we have to `Nifti`?

choldgraf, 2017-01-27:
This is not really an MNE-specific conversation, but I think it might be relevant to folks in this world so I figured here is as good a place as any to ask it.

What's the closest thing we've got in electrophysiology to a `nifti` file? I feel like I've seen a few projects popping up lately where people are trying to sort of recreate the wheel. For example, right now I know of:

* `.fif` format, which MNE uses
* `.nwb` format, which is being developed by neurodata without borders and is more domain-general
* `.brain` format, which is being developed out of LBNL and is (maybe?) somewhat related to neurodata without borders
* `.edf` format, which is a fairly common data standard but I rarely see people storing their data in this format
* `.XXX` a random smattering of a bazillion other formats for electrophysiology data.

I was looking through the NWB project in particular, and was a little frustrated at the fact that we don't really have something like `nifti` that sets some kind of standard for electrophysiology. I think there are a lot of differences between ephys and fMRI (e.g. we have way more types of acquisition systems than there are fMRI machines), but is having some kind of common I/O format just an unsolvable problem in ephys? I'm not sure, but I'm curious what the MNE community thinks about this.

Any thoughts? I'm just curious to know if others have thought about this. Or if this is just totally off-topic I can close the issue ;-)

agramfort, 2017-01-27:
my opinion is "choose your battles" :)

choldgraf, 2017-01-27:
haha fair - I think I agree in general. I just keep seeing people trying to re-create their own file format in their own silo'd worlds etc. The thing is, I feel like when this happens a lot it means that either:

A. There should be some kind of common file structure because people keep inventing it for their workflows, or
B. There definitely shouldn't be some kind of common file structure because everybody seems to have opinions about how they want it done

dgwakeman, 2017-01-27:
To expand a bit on this. I too felt the same way and did quite a bit of lobbying. A while later, I discovered just what a disaster nifti is as a format. One nifti doesn't match another if a different converter is used (in fact the most popular create some of the worst (with little details in the header). Plus the confines of nifti means really important information doesn't fit in the header (timing of slices; b-vectors, etc.). Now there is a new project called bids to try to deal with some of these issues.

I think putting those ideas into making the formats mne uses as useable as possible is the best way, and if you collaborate in other projects to help them read your files which have features that are necessary for whatever cool new techniques you do. Only necessary data will ever get added as a feature in a project.

choldgraf, 2017-01-27:
that's a good point @dgwakeman. I should say I am not lobbying for having a common format :) I mostly just want to see what people think about it because it's a complex issue.

I was under the impression that `bids` format was sort of operating at a higher level than nifti...that it was meant to be a way to organize entire experiments / meta-data / etc. Maybe I'll look into it more.

I was reminded of this problem because I was looking through the neurodata without borders project, and they have a lot of work to do on their API. One challenge is that it seems like they're sort of trying to solve the "low-level I/O file format" problem at the same time as the "high level reproducible science w/ entire file structures and workflows" problem. This seems really hard to do in a way that will satisfy users.

choldgraf, 2017-01-27:
E.g., one question that's been popping up in my head is whether it's worth having a file type built on hdf5. It seems like a lot of the file formats for data are moving in this direction and it may make it easier for people to read / understand the data structures without necessarily needing a "reader function" in case they don't have one.

agramfort, 2017-01-27:
FYI there is a BIDS format for MEG on the way.

see recent efforts

https://github.com/brainstorm-tools/bst-data-examples

choldgraf, 2017-01-27:
very cool! I haven't used BIDS much myself so I don't know how well-designed it is, but I've heard good things!

larsoner, 2018-04-04:
Closing for BIDS

closed 2018-04-04 by larsoner